v-zhuravlev/zbx-smartctl #10: Make items and UserParams compatible with dev names like "/dev/csmi0,0"

lisikh, 2016-03-24:
I was getting errors with devices called "/dev/csmi0,0", ""/dev/csmi0,1" etc on Windows hosts.
To fix it:

- Used double-quotes on "$1" params in UserParams:
```
UserParameter=uHDD[*], for /F "tokens=10" %a in ('C:\usr\zabbix\smartmontools\bin\smartctl.exe -A "$1" ^| find "$2"') do @echo %a
UserParameter=uHDD.health.[*], for /F "tokens=6" %a in ('C:\usr\zabbix\smartmontools\bin\smartctl.exe -H "$1" ^| find "test"') do @echo %a
UserParameter=uHDD.model.[*],for /F "tokens=3*" %a in ('C:\usr\zabbix\smartmontools\bin\smartctl.exe -i "$1" ^| find "Device Model"') do @echo %a %b
UserParameter=uHDD.sn.[*],for /F "tokens=3" %a in ('C:\usr\zabbix\smartmontools\bin\smartctl.exe -i "$1" ^| find "Serial Number"') do @echo %a
UserParameter=uHDD.errorlog.[*], for /F "tokens=4" %a in ('C:\usr\zabbix\smartmontools\bin\smartctl.exe -l error "$1" ^| find "ATA Error Count"') do @echo %a
```

- Escaped macro {#DISKNAME} in calcuted item "{#DISKNAME}: SMART critical errors total":

```
last("uHDD[\"{#DISKNAME}\",5 Reallocated]")+last("uHDD[\"{#DISKNAME}\",Spin_Retry_Count]")+last("uHDD[\"{#DISKNAME}\",Current_Pending_Sector]")+last("uHDD[\"{#DISKNAME}\",Offline_Uncorrectable]")
```

Also, you ps-discovery script looks only into devices with name likes "/dev/hd*". 
So I have written bat-file, which calls "smartctl.exe --scan" to detect hard drives.
[smartctl-disks-discovery.bat.zip](https://github.com/v-zhuravlev/zbx-smartctl/files/188046/smartctl-disks-discovery.bat.zip)

v-zhuravlev, 2016-05-26:
@lisikh , thanks for contributing. I modified your script a bit: ```

v-zhuravlev, 2016-05-26:
also scan found two devices in my laptop under System: sda, csmi0,0 both point to the same HDD:

```
C:\Program Files (x86)\Zabbix Agent>smartctl-disks-discovery.bat
{"data":[
{"{#DISKNAME}":"/dev/sda","{#SMART_ENABLED}":"1"},
{"{#DISKNAME}":"/dev/csmi0,0","{#SMART_ENABLED}":"1"}
]}

C:\Program Files (x86)\Zabbix Agent>smartctl -i /dev/csmi0,0
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-win7(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Samsung based SSDs
Device Model:     Samsung SSD 850 EVO 500GB
Serial Number:    S21HNXAG464372R
LU WWN Device Id: 5 002538 d4008d842
Firmware Version: EMT01B6Q
User Capacity:    500 107 862 016 bytes [500 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2, ATA8-ACS T13/1699-D revision 4c
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu May 26 17:03:46 2016 RTZ
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

C:\Program Files (x86)\Zabbix Agent>smartctl -i /dev/sda
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-win7(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Samsung based SSDs
Device Model:     Samsung SSD 850 EVO 500GB
Serial Number:    S21HNXAG464372R
LU WWN Device Id: 5 002538 d4008d842
Firmware Version: EMT01B6Q
User Capacity:    500 107 862 016 bytes [500 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2, ATA8-ACS T13/1699-D revision 4c
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu May 26 17:04:05 2016 RTZ
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```

lisikh, 2016-05-27:
@v-zhuravlev , yes, your version also works for me.
My test was founded on SSHD drives, scan also shows them with different names, but:
```
c:\Program Files\Zabbix\scripts\prisma>c:\zabbix\smartctl.exe -i /dev/sda
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: Input/output error

A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.
```
/dev/csmi0* works good:
```
c:\Program Files\Zabbix\scripts\prisma>c:\zabbix\smartctl.exe -i /dev/csmi0,0
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Desktop SSHD
Device Model:     ST4000DX001-1CE168
Serial Number:    Z304GWTJ
LU WWN Device Id: 5 000c50 07bb7577c
Firmware Version: CC45
User Capacity:    4 000 787 030 016 bytes [4,00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5900 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2, ACS-3 T13/2161-D revision 3b
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Fri May 27 13:48:38 2016 RTZST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```
And if I want /dev/sda to display properly, I need to specify "-d scsi" option. Though it works only with some options:
```
c:\Program Files\Zabbix\scripts\prisma>c:\zabbix\smartctl.exe -i /dev/sda -d scsi
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               ST4000DX
Product:              001-1CE168
Revision:             CC45
User Capacity:        4 000 787 030 016 bytes [4,00 TB]
Logical block size:   512 bytes
Physical block size:  4096 bytes
Rotation Rate:        5900 rpm
Form Factor:          3.5 inches
Logical Unit id:      0x5000c5007bb7577c
Serial number:        Z304GWTJ
Device type:          disk
Local Time is:        Fri May 27 13:48:46 2016 RTZST
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Disabled or Not Supported

c:\Program Files\Zabbix\scripts\prisma>c:\zabbix\smartctl.exe -A /dev/sda -d scsi
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===

```
So, in this case by default I get only one device with smart enabled.

v-zhuravlev, 2016-05-30:
Here is the new version that removes duplicates (by Serial number)
and appends -d type to the name so disk can be properly accessed.
https://raw.githubusercontent.com/v-zhuravlev/zbx-smartctl/421d61748d9281a59c201c211238386b1cd82eb9/discovery-scripts/windows/smartctl-disks-discovery.bat

lisikh, 2016-06-01:
It works for me as your previous variant.
**smartctl --scan** shows "-d ata" for all devices, so it cannot retreive general info from duplicates and I'm  getting {#SMART_ENABLED}=0 for such entries (which is ok).

v-zhuravlev, 2016-06-01:
I didn't quite get it. Can you provide the script output as well `smartctl --scan` output?

lisikh, 2016-06-01:
Here it is:
```

C:\Users\Administrator\Desktop>c:\zabbix\smartctl.exe --scan 
/dev/sda -d ata # /dev/sda, ATA device
/dev/sdb -d ata # /dev/sdb, ATA device
/dev/sdc -d ata # /dev/sdc, ATA device
/dev/csmi0,0 -d ata # /dev/csmi0,0, ATA device
/dev/csmi0,1 -d ata # /dev/csmi0,1, ATA device
/dev/csmi0,5 -d ata # /dev/csmi0,5, ATA device

C:\Users\Administrator\Desktop>c:\zabbix\smartctl.exe -i /dev/sda -d ata 
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: Input/output error

A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.

C:\Users\Administrator\Desktop>c:\zabbix\smartctl.exe -i /dev/sda -d scsi 
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               ST4000DX
Product:              001-1CE168
Revision:             CC45
User Capacity:        4 000 787 030 016 bytes [4,00 TB]
Logical block size:   512 bytes
Physical block size:  4096 bytes
Rotation Rate:        5900 rpm
Form Factor:          3.5 inches
Logical Unit id:      0x5000c5007bb7577c
Serial number:        Z304GWTJ
Device type:          disk
Local Time is:        Wed Jun 01 16:33:57 2016 RTZST
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Disabled or Not Supported

C:\Users\Administrator\Desktop>c:\zabbix\smartctl.exe -i /dev/csmi0,0 -d ata 
smartctl 6.4 2015-06-04 r4109 [i686-w64-mingw32-2008r2(64)-sp1] (sf-6.4-1)
Copyright (C) 2002-15, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Desktop SSHD
Device Model:     ST4000DX001-1CE168
Serial Number:    Z304GWTJ
LU WWN Device Id: 5 000c50 07bb7577c
Firmware Version: CC45
User Capacity:    4 000 787 030 016 bytes [4,00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5900 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2, ACS-3 T13/2161-D revision 3b
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Wed Jun 01 16:33:57 2016 RTZST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```

mykolq, 2016-08-04:
Hm, i have en empty fields  value. What can be wrong? Discovery scripts from here [https://raw.githubusercontent.com/v-zhuravlev/zbx-smartctl/421d61748d9281a59c201c211238386b1cd82eb9/discovery-scripts/windows/smartctl-disks-discovery.bat](url) . Zabbix parameters from main page. 
```
C:\zabbix_agent\scripts>C:\smartmontools\bin\smartctl.exe --scan
/dev/sda -d ata # /dev/sda, ATA device
/dev/csmi0,0 -d ata # /dev/csmi0,0, ATA device
```
```
C:\zabbix_agent\scripts>C:\smartmontools\bin\smartctl.exe -i /dev/csmi0,0 -d ata
smartctl 6.5 2016-05-07 r4318 [x86_64-w64-mingw32-win10] (sf-6.5-1)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     SAMSUNG MZMTD128HAFV-000
Serial Number:    S15MNYAD439402
LU WWN Device Id: 5 002538 500000000
Firmware Version: DXT41K0Q
```
but
`C:\zabbix_agent\scripts>for /F "tokens=3*" %a in ('C:\smartmontools\bin\smartctl.exe -i "/dev/csmi0,0 -d ata" ^| find "Device Model"') do @echo %a %b`

`C:\zabbix_agent\scripts>for /F "tokens=3*" %a in ('C:\smartmontools\bin\smartctl.exe -i /dev/csmi0,0 -d ata ^| find "Device Model"') do @echo %a %b
`

Nemles, 2016-08-17:
Если использовать батник в дискавери вместо PS, то Value should be a JSON object.

Вывод батника такой: 
{"data":[
{"{#DISKNAME}":"/dev/sda -d scsi","{#SMART_ENABLED}":"0"}
,
{"{#DISKNAME}":"/dev/sdb -d scsi","{#SMART_ENABLED}":"0"}
,
{"{#DISKNAME}":"/dev/sdc -d sat","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/sdd -d sat","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,0 -d ata","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,1 -d ata","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,2 -d ata","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,3 -d ata","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,4 -d ata","{#SMART_ENABLED}":"0"}
]}

Проверку на http://jsonlint.com/ проходит.
Заббикс 3.0.

mykolq, 2016-08-19:
Да батник то работает, а вот в заббикс в значениях нули =)

Nemles, 2016-08-19:
да я к тому что дискавери не отрабатывает у меня с ним)

mykolq, 2016-08-19:
ну да =)

mykolq, 2016-10-11:
в общем вернулся я к той же проблеме? ни у кого так и не получилось?

v-zhuravlev, 2016-10-12:
Emn, @lisikh , @mykolq  lets start from over again.

What happens when you specify `smartctl -i /dev/sda -A` or `smartctl -i /dev/csmi0,0 -A` while running cli under Admin rights without actually specifying -d key? 

1. Could you summarize which smartctl commands fail and which does not?

2. I think adding things like `-d sata` to the macros isn't an elegant idea, so I would probably would try to avoid it at the end.
3. It appears to be that `csmi0,0` and `sda` are actually pointing to the same device. I have no problems to access disk data using /dev/sda without specifying -d key. Is that true that you can access disk data only by `csmi0,0`?

mykolq, 2016-10-19:
Hello, @v-zhuravlev 
`C:\WINDOWS\system32>smartctl -i /dev/sda -A
smartctl 6.5 2016-04-12 r4290 [x86_64-w64-mingw32-win10] (daily-20160412)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: Input/output error

A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.
`
and 
`C:\WINDOWS\system32>smartctl -i /dev/sda -A
smartctl 6.5 2016-04-12 r4290 [x86_64-w64-mingw32-win10] (daily-20160412)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: Input/output error

A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.
`

mykolq, 2016-12-16:
@Nemles @v-zhuravlev забыл отписаться, все работает. в батнике строку `echo {"{#DISKNAME}":"%%i %%j %%k,"{#SMART_ENABLED}":"!smart_enabled!"}` привести к виду `echo {"{#DISKNAME}":"%%i,"{#SMART_ENABLED}":"!smart_enabled!"}`.  smartctl 6.5

It works with `echo {"{#DISKNAME}":"%%i,"{#SMART_ENABLED}":"!smart_enabled!"}` instead of  `echo {"{#DISKNAME}":"%%i %%j %%k,"{#SMART_ENABLED}":"!smart_enabled!"}` in bat file.

Nemles, 2016-12-16:
@mykolq Спс, а подскажи какая версия у тебя сервака заббикса и клиента на винде?
И можешь тогда батник  свой показать полностью, потому что у меня все также не валидные данные json, я с тех пор не ковырял это.

mykolq, 2016-12-16:
@Nemles `@echo off
setlocal enabledelayedexpansion

set smartctl_path=C:\smartmontools\bin\smartctl.exe
set serials=dummy_serial
set /a first=1

echo {"data":[
for /f "tokens=1,2,3 delims= " %%i in ('!smartctl_path! --scan') do (
	set smart_enabled=0
    set duplicate=0
    for /F "tokens=3*" %%a in ('!smartctl_path! -i %%i %%j %%k^| find "Serial Number"') do (
        set serial=%%a
    )

        rem duplicate disk entity, check by serial number
        for %%j in (!serials!) do (if %%j == !serial! (set /a duplicate+=1)) 
        set serials=%serials% !serial!

    
	FOR /F "tokens=1" %%a IN ('!smartctl_path! -i %%i %%j %%k ^|find "SMART" ^| find /C "Enabled"') DO (
		if %%a gtr 0 (
        set /a smart_enabled=1
        )
	)
    if not !duplicate! == 1 (
            if not !first! ==1 (
                echo ,
            )
            echo {"{#DISKNAME}":"%%i","{#SMART_ENABLED}":"!smart_enabled!"}
            set /a first=0
    )
)
echo ]}`
Zabbix server 3.2.2, клиенты пока что 3.0.1

Knorkator, 2017-04-06:
With your Base-Powershell-Script, csmi Devices like Raid-0 HDD are not shown.
The script here recognizes csmi controller and uses the found hard disks if available.
Please excuse the English.
I am not a programmer, so please be tolerant
:)

`
$script:smartctl = "C:\Programme\smartmontools\bin\smartctl.exe" 

function GetSMARTDevice()
{    
    $scan = & $smartctl "--scan" 
    
    
    if ($scan -like "/dev/csmi*") # For Intel Onboard Controllers
    {
        $scan=$scan | Select-String -Pattern '/dev/csmi*' 
        $script:scan=$scan -replace ".-d.*"        
        $script:AnzahlGeraete=$scan.count
    }
    elseif ($scan -like "/dev/sd*") # The other Controllers - At this moment.. maybe there are more Controllertypes.. :)
    {
        $scan=$scan | Select-String -Pattern '/dev/sd*'
        $script:scan=$scan -replace ".-d.*"
        $script:AnzahlGeraete=$scan.count
    }

}

function createjson($disks,$DeviceCount)
{
    $idx =1
     

    write-host "{"
    write-host " `"data`":[`n"

    Foreach ($smartctl_disk_name in $disks)
    {
          #Device-Check
            # If SMART is Enabled, $smart_enabled will be 1.
            # The Value will be used in the Discovery Rule            
            $smart_enabled = & $smartctl "-i" $smartctl_disk_name | select-string "SMART.+Enabled$" 
            if($smart_enabled) {$smart_enabled = 1} 
            else {$smart_enabled = 0}
  
            # Is it HDD, SSD or ODD
            # The SMART Values for HDD/SSD are different sometimes
            # I have 2 Discovery-Rules with Filtering 0 or 1.                       
            # 0 is for HDD
            # 1 is for SSD
            # 2 is for ODD and will be ignored
            $Drive = & $smartctl "-i" $smartctl_disk_name  | select-string "Rotation Rate:" 
            if($Drive -like "*Solid State Device*") {$DriveType = "1"} 
            elseif ($Drive -like "*rpm*") {$DriveType = "0"}
            else { $DriveType = "2"}
            
    
            # Device Name
            # If there is a device-description, it will be used
            # If not, the /dev/... Value will be used
            $model=& $smartctl "-a" $smartctl_disk_name | select-string "Device Model:"
            $model=$model -replace "Device Model:"
            $model=$model.trim()            
            if($model) {$model = "$model" } 
            else {$model = "$smartctl_disk_name"}

          #JSON 
            if ($idx -lt $DeviceCount) 
            {                
                $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +",`n" + "`t`t`"{#SSDODERHDD}`":`""+$DriveType+"`"" +",`n" + "`t`t`"{#DEVICENAME}`":`""+$Model+"`"" +"`n`t},`n"        
                write-host $line
            }

            if ($idx -ge $DeviceCount)
            {
                $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +",`n" + "`t`t`"{#SSDODERHDD}`":`""+$DriveType+"`"" +",`n" + "`t`t`"{#DEVICENAME}`":`""+$Model+"`"" +"`n`t}`n"        
                write-host $line
            }
                $idx++;

    } # End of Foreach

    write-host
    write-host " ]"
    write-host "}"    

} # End Of Function

getsmartdevice
createjson -disks $scan -DeviceCount $AnzahlGeraete`

hyperion11, 2017-12-18:
use @Knorkator script
work fine, except some parametes with latest zabbix server 3.4.4 and agent 3.4.2 and latest smartmontools-6.6-1.win32
screens included
[image]

[image]

[image]

hyperion11, 2018-02-14:
Использую скрипт @Knorkator 
Оказывается не передается почему то CSMI0,0
В JSON он попадает от скрипта
```
PS C:\Windows\system32> C:\Program Files\Zabbix Agent\smartctl-disks-discovery.ps1
{
 "data":[

	{
 		"{#DISKNAME}":"/dev/csmi0,0",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"1",
		"{#DEVICENAME}":"KINGSTON SA400S37240G"
	},

	{
 		"{#DISKNAME}":"/dev/csmi0,1",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"1",
		"{#DEVICENAME}":"KINGSTON SKC300S37A240G"
	},

	{
 		"{#DISKNAME}":"/dev/csmi0,2",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"0",
		"{#DEVICENAME}":"WDC WD1003FBYX-01Y7B1"
	},

	{
 		"{#DISKNAME}":"/dev/csmi0,3",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"0",
		"{#DEVICENAME}":"WDC WD1003FBYX-01Y7B1"
	},

	{
 		"{#DISKNAME}":"/dev/csmi0,4",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"0",
		"{#DEVICENAME}":"WDC WD2003FYYS-02W0B1"
	},

	{
 		"{#DISKNAME}":"/dev/csmi0,5",
		"{#SMART_ENABLED}":"1",
		"{#SSDODERHDD}":"0",
		"{#DEVICENAME}":"WDC WD2003FYYS-02W0B1"
	}

 ]
}
```
а вот сам zabbix нулевой канал почему то не видит
[image]

hyperion11, 2018-02-15:
все работает. Это я дурак, зачем то отключил в настройках сервера этот параметр :)

agatus, 2018-02-26:
PS C:\> powershell -NoProfile -ExecutionPolicy Bypass -File 1.ps1
{
 "data":[

Не удается преобразовать значение "2`" в тип "System.Int32". Ошибка: "Входная строка имела неверный формат."
C:\1.ps1:61 знак:13
+         if ($idx -lt $DeviceCount)
+             ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvalidCastFromStringToInteger

Не удается преобразовать значение "2`" в тип "System.Int32". Ошибка: "Входная строка имела неверный формат."
C:\1.ps1:67 знак:13
+         if ($idx -ge $DeviceCount)
+             ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvalidCastFromStringToInteger

Не удается преобразовать значение "2`" в тип "System.Int32". Ошибка: "Входная строка имела неверный формат."
C:\1.ps1:61 знак:13
+         if ($idx -lt $DeviceCount)
+             ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvalidCastFromStringToInteger

Не удается преобразовать значение "2`" в тип "System.Int32". Ошибка: "Входная строка имела неверный формат."
C:\1.ps1:67 знак:13
+         if ($idx -ge $DeviceCount)
+             ~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvalidCastFromStringToInteger

 ]
}

C:\Program Files\smartmontools\bin>smartctl.exe --scan
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d scsi # /dev/sdb, SCSI device
/dev/sdc -d sat # /dev/sdc [SAT], ATA device
/dev/csmi1,3 -d ata # /dev/csmi1,3, ATA device
/dev/csmi2,0 -d ata # /dev/csmi2,0, ATA device

hyperion11, 2018-02-26:
PS запущена с правами администратора? ПКМ - "запуск от имени администратора"

agatus, 2018-02-26:
Да, конечно с правами админа.

hyperion11, 2018-02-27:
@agatus 
[image]

на всякий случай код скрипта такой
`
$script:smartctl = "C:\Program Files\smartmontools\bin\smartctl.exe"

function GetSMARTDevice()
{
$scan = & $smartctl "--scan"

if ($scan -like "/dev/csmi*") # For Intel Onboard Controllers
{
    $scan=$scan | Select-String -Pattern '/dev/csmi*' 
    $script:scan=$scan -replace ".-d.*"        
    $script:AnzahlGeraete=$scan.count
}
elseif ($scan -like "/dev/sd*") # The other Controllers - At this moment.. maybe there are more Controllertypes.. :)
{
    $scan=$scan | Select-String -Pattern '/dev/sd*'
    $script:scan=$scan -replace ".-d.*"
    $script:AnzahlGeraete=$scan.count
}
}

function createjson($disks,$DeviceCount)
{
$idx =1

write-host "{"
write-host " `"data`":[`n"

Foreach ($smartctl_disk_name in $disks)
{
      #Device-Check
        # If SMART is Enabled, $smart_enabled will be 1.
        # The Value will be used in the Discovery Rule            
        $smart_enabled = & $smartctl "-i" $smartctl_disk_name | select-string "SMART.+Enabled$" 
        if($smart_enabled) {$smart_enabled = 1} 
        else {$smart_enabled = 0}

        # Is it HDD, SSD or ODD
        # The SMART Values for HDD/SSD are different sometimes
        # I have 2 Discovery-Rules with Filtering 0 or 1.                       
        # 0 is for HDD
        # 1 is for SSD
        # 2 is for ODD and will be ignored
        $Drive = & $smartctl "-i" $smartctl_disk_name  | select-string "Rotation Rate:" 
        if($Drive -like "*Solid State Device*") {$DriveType = "1"} 
        elseif ($Drive -like "*rpm*") {$DriveType = "0"}
        else { $DriveType = "2"}
        

        # Device Name
        # If there is a device-description, it will be used
        # If not, the /dev/... Value will be used
        $model=& $smartctl "-a" $smartctl_disk_name | select-string "Device Model:"
        $model=$model -replace "Device Model:"
        $model=$model.trim()            
        if($model) {$model = "$model" } 
        else {$model = "$smartctl_disk_name"}

      #JSON 
        if ($idx -lt $DeviceCount) 
        {                
            $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +",`n" + "`t`t`"{#SSDODERHDD}`":`""+$DriveType+"`"" +",`n" + "`t`t`"{#DEVICENAME}`":`""+$Model+"`"" +"`n`t},`n"        
            write-host $line
        }

        if ($idx -ge $DeviceCount)
        {
            $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +",`n" + "`t`t`"{#SSDODERHDD}`":`""+$DriveType+"`"" +",`n" + "`t`t`"{#DEVICENAME}`":`""+$Model+"`"" +"`n`t}`n"        
            write-host $line
        }
            $idx++;

} # End of Foreach

write-host
write-host " ]"
write-host "}"    
} # End Of Function

getsmartdevice
createjson -disks $scan -DeviceCount $AnzahlGeraete
`

agatus, 2018-02-27:
Может потому что это windows server 2012?

hyperion11, 2018-02-27:
хм, в 2008 ругается на trim()
[image]

agatus, 2018-02-27:
У меня в 2008R2 отлично отрабатывает.

hyperion11, 2018-02-27:
какая версия PS?
[image]

agatus, 2018-02-27:
на 2008 - 2.0
на 2012 - 3.0

hyperion11, 2018-02-27:
на 4.0 все работает. На 3.0 выдает ошибку.  У вас на 3.0 тоже ошибка?

agatus, 2018-02-27:
да, но не на трим ругается. Думаете это из-за версии шела?

hyperion11, 2018-02-27:
Только догадки..

tommer-cpu, 2025-06-25:
> Я не совсем понял. Можете ли вы предоставить вывод скрипта, а также `smartctl --scan`вывод?

Уважаемый v-zhuravlev,простите меня за тупость можно с вами как то связатся мне нужна ваша помошь,вот моя почта pchealing@gmail.com