v-zhuravlev/zbx-smartctl #14: Discovery script for *BSD

mattpdmnz, 2016-05-23:
Hi

You can turn the Linux discovery script into a BSD-compatible script (particularly useful for FreeNAS and similar) with the following change:
- for (`ls -l /dev/disk/by-id/ | cut -d"/" -f3 | sort -n | uniq -w 3`)
+ for (`ls /dev/ | egrep -w \'(a|s|w)?da?[0-9]+$\'`)

This is portable across FreeBSD, NetBSD and OpenBSD.

Hope you find this useful.

v-zhuravlev, 2016-05-23:
Thanks for posting. It doesn't work in in linux for me though :
```
ls /dev/ | egrep -w \'(a|s|w)?da?[0-9]+$\
-bash: syntax error near unexpected token `('
```

mattpdmnz, 2016-05-25:
The '' is escaped to run in perl. You need to remove the \ to run from the shell. And on Linux it will only return sda and partitions, eg: sda, sda1, sda2. BSD drives are labelled quite differently.

I was only writing it for BSD discovery to go along with your Linux script. You could do an environment test with "uname -o" and then use one of the follow based on the result, to turn it into a truly portable script. I don't know perl so cannot help you with that testing. Have included OS X (Darwin), too, so you could replace your separate OS X script.

BSD: ls /dev/ | egrep -w '(a|s|w)?da?[0-9]+$'
Linux: ls /dev/ | egrep -w '(h|s)d[a-z]+$'
Darwin: ls /dev/ | egrep -w 'disk[0-9]+$'

Hopefully that makes sense.

v-zhuravlev, 2016-05-26:
tell me, would command `smartctl --scan` work? what would be the output for BSD?

mattpdmnz, 2016-05-26:
From a FreeNAS (FreeBSD) box:
`smartctl --scan`
`/dev/da0 -d scsi # /dev/da0, SCSI device`
`/dev/da1 -d scsi # /dev/da1, SCSI device`
`/dev/da2 -d scsi # /dev/da2, SCSI device`

Same format as Linux, so you can just cut the first field and use that directly. 

It **doesn't** work usefully on OS X, but that can be worked around.

v-zhuravlev, 2016-05-27:
Thanks for info,
Could you test the latest script with freenas, would it produce a proper JSON?
https://github.com/v-zhuravlev/zbx-smartctl/commit/cef5bc7cd29ebb7aa9461c8775fe386eaf6de1ff

mattpdmnz, 2016-05-29:
Looks good.
```
perl smartctl-disks-discovery.pl
{
        "data":[

                {
                        "{#DISKNAME}":"",
                        "{#SMART_ENABLED}":"0"
                },
                {
                        "{#DISKNAME}":"da0",
                        "{#SMART_ENABLED}":"1"
                },
...
                        "{#DISKNAME}":"ada2",
                        "{#SMART_ENABLED}":"1"
                }
        ]
}

```

v-zhuravlev, 2016-05-30:
It doesn't match the output from smartlctl --scan you posted earlier for some reason though...

v-zhuravlev, 2016-05-30:
I wonder why there are `...` And empty disk name

mattpdmnz, 2016-05-30:
The … was me trimming out a lot of disks. The output looked correct.
The drive without a name was due to my server having a missing default entry in its smartctl database. The output from smartctl --scan looks like:

```
smartctl --scan
Warning: DEFAULT entry missing in drive database file(s)
/dev/da0 -d scsi # /dev/da0, SCSI device
/dev/da1 -d scsi # /dev/da1, SCSI device
/dev/da2 -d scsi # /dev/da2, SCSI device
/dev/da3 -d scsi # /dev/da3, SCSI device
/dev/da4 -d scsi # /dev/da4, SCSI device
/dev/da5 -d scsi # /dev/da5, SCSI device
/dev/da6 -d scsi # /dev/da6, SCSI device
/dev/da7 -d scsi # /dev/da7, SCSI device
/dev/da8 -d scsi # /dev/da8, SCSI device
/dev/da9 -d scsi # /dev/da9, SCSI device
/dev/da10 -d scsi # /dev/da10, SCSI device
/dev/da11 -d scsi # /dev/da11, SCSI device
/dev/ses0 -d scsi # /dev/ses0, SCSI device
/dev/ada0 -d atacam # /dev/ada0, ATA device
/dev/ada1 -d atacam # /dev/ada1, ATA device
/dev/ada2 -d atacam # /dev/ada2, ATA device
```
Your script appears to be fine. You could filter the output of the scan for lines ending with "device" to avoid it trying to parse error messages, but that's about it.

v-zhuravlev, 2016-10-12:
@mattpdmnz , If you are still here, could you please check commit https://github.com/v-zhuravlev/zbx-smartctl/commit/39c317bf71ee8da919815a16fcf48c310eb3270c ? Warning line should be filtered out now

PR #20 (2016-10-14): added smartctl output basic validation 
(to filter out lines like 'Warning: DEFAULT entry missing in drive database file(s)' in #14
Fixes #14
closed 2016-10-14 by v-zhuravlev