v-zhuravlev/zbx-smartctl #21: debhelper install file is incorrect

alllel, 2017-02-03:
1. Syntax of _install_ file is: source_file destination_DIR, not dest_FILE
2. On Ubuntu, default directory for zabbix_agent configs is /etc/zabbix/zabbix_agentd.conf.d not just /etc/zabbix/zabbix_agentd.d

Proposed file zabbix-agent-extra-smartctl.install :
>  sudoers_zabbix_smartctl etc/sudoers.d
>  discovery-scripts/nix/smartctl-disks-discovery.pl usr/lib/zabbix/scripts
>  zabbix_smartctl.conf etc/zabbix/zabbix_agentd.conf.d

v-zhuravlev, 2017-02-03:
@vazhnov , could you please comment on this? Thank you

alllel, 2017-02-03:
One more thing: script should be done executable somehow.
Now it is packaged into deb as non-executable, so discovery does not work.

alllel, 2017-02-03:
And one more thing: package should have postinst/postrm files to reload zabbix-agent on installation/deinstallation.
Something like that: debian/zabbix-agent-extra-smartctl.postinst
>#!/bin/sh
>
>set -e
>
>case $1 in 
>    configure)
>	invoke-rc.d zabbix-agent restart
>esac
>
>#DEBHELPER#
>

commit b9cbee16 (2017-02-03): better packaging (#22)

fixes #21 .Thanks for contributing
closed 2017-02-03 by v-zhuravlev