v-zhuravlev/zbx-smartctl #30: SMART_ENABLED is always '0' in Debian 6

perederyaev, 2017-05-18:
Hi!
I have Debian 6 and smartctl 5.40

$ smartctl -i /dev/sda
smartctl 5.40 2010-07-12 r3124 [i686-pc-linux-gnu] (local build)
Copyright (C) 2002-10 by Bruce Allen, http://smartmontools.sourceforge.net

=== START OF INFORMATION SECTION ===
......
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

$ /etc/zabbix/scripts/smartctl-disks-discovery.pl
{
        "data":[
                {
                        "{#DISKNAME}":"/dev/sda -d sat",
                        "{#SMART_ENABLED}":"0"
                }
        ]
}

# smartctl --scan-open
/dev/sda [SAT] -d sat [ATA] (opened)
/dev/sdb [SAT] -d sat [ATA] (opened)

To make it work I had to change line 21 of script /etc/zabbix/scripts/smartctl-disks-discovery.pl:
from
$disk = "@device[0] @device[1] @device[2]";
to
$disk = "@device[0] @device[2] @device[3]";

Thank You!

shadbarton, 2017-05-18:
Hi
Debian 8
smartctl 6.4 2014-10-07 
smartctl --scan-open
/dev/sda -d sat # /dev/sda [SAT], ATA device
/dev/sdb -d sat # /dev/sdb [SAT], ATA device
smartmontools version may cause your error?

v-zhuravlev, 2017-05-22:
@perederyaev, hi try this fix: https://github.com/v-zhuravlev/zbx-smartctl/commit/7453390fe28e327a2b9d7944c9445c0d80b9222b

commit bd7edd1e (2017-05-23): to fix #30 (and hopefully not to break anything else) (#31)
closed 2017-05-23 by v-zhuravlev