v-zhuravlev/zbx-smartctl #36: SSD/HDD different LLD

v-zhuravlev, 2017-09-02:
"Под Linux, оказывается, можно всего одной командой определить какой диск - SSD или HDD:

Linux automatically detects SSD, and since kernel version 2.6.29, you may verify sda with:
cat /sys/block/sda/queue/rotational
You should get 1 for hard disks and 0 for a SSD.

То есть, можно добавить в perl-script еще одну проверку с всего одной командой, вывод которой даст определение типа диска - 1 или 0.

Далее добавить еще один RegExp "DISK_TYPE". Если он - "1", то ничего делать не надо. Если он "0" и имеем дело с SSD, то надо делать еще одно Правило обнаружения с элементами данных под SSD. Там самое главное - это 231-й параметр вывода smartctl, который и показывает "Здоровье диска". Из остальных параметров можно оставить 5, 9, 194 (надо редактировать), инвентарные данные, а так же Test Result. Все остальное не актуально для SSD.

Вот только, насколько я знаю, несколько RegExp в Правилах обнаружения Zabbix умеет в версии 3+. На 2.2 и 2.4 придется играться с глобальными регулярками, и писать правила туда. Для HDD это будет "SMART_ENABLED = 1 и DISK_TYPE = 1", для SSD - "SMART_ENABLED = 1 и DISK_TYPE = 0"

В общем, все можно красиво реализовать. И было бы просто потрясающе сделать модернизацию шаблона.

Спасибо! :)"

vazhnov, 2017-09-21:
As I know, 231 is not for any SSD manufactures:
* Corsair with SandForce — 231 SSD_Life_Left,
* Crucial with Micron — 173 Ave_Block-Erase_Count, 202 Percent_Lifetime_Used
* Intel — 233 Media_Wearout_Indicator

Do you know another types?

v-zhuravlev, 2017-09-21:
Samsung - 177

v-zhuravlev, 2017-09-25:
"Corsair with SandForce — 231 SSD_Life_Left,
Crucial with Micron — 173 Ave_Block-Erase_Count, 202 Percent_Lifetime_Used"

do those have 233? Do you have example output you can share for those disks?

v-zhuravlev, 2018-05-24:
added #DISKTYPE macro with values 0 -HHD, 1-SSD, 0-Other. this now can be used for separate discovery rules. Not sure whether it should be separated though in the templates stored here.

closed 2018-05-24 by v-zhuravlev