v-zhuravlev/zbx-smartctl #44: it would be nice if the discovery have more options

colttt, 2017-11-03:
Hello,

at first, great job ;)
BUT, I've a JBOD with al lot disks (42) and i use partlabel to identify this with a schema, for example j1d01-ssd that means in jbod1 disk 01 and its an ssd, so it would be nice if i see that in zabbix instead of sdam, also some users with a lot of disks are using by-uuid and get pci-0000:03:00.0-sas-exp0x50030480091018ff-phy3-lun-0 

best regards

v-zhuravlev, 2017-11-03:
HI, thanks
- can you give the output of discovery?
- and some sample output of `smartctl`?
- and what do you want to replace with what again.

colttt, 2017-11-03:
Hi,

- can you give the output of discovery script?
```
perl smartctl-disks-discovery.pl 
{
        "data":[

                {
                        "{#DISKNAME}":"/dev/sda -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdb -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdc -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdd -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sde -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdf -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdg -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdh -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdi -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdj -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdk -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdl -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdm -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdn -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdo -d scsi",
                        "{#SMART_ENABLED}":"1"
                },
                {
                        "{#DISKNAME}":"/dev/sdp -d scsi",
[... 303 lines cut ...]
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy32-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy33-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy33-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy34-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy34-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy35-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy35-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy36-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy36-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy3-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy3-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy4-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy4-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy5-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy5-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy6-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy6-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy7-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy7-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy8-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy8-lun-0-part1
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy9-lun-0
pci-0000:03:00.0-sas-exp0x50030480091018ff-phy9-lun-0-part1
```
thanks in advance

v-zhuravlev, 2017-11-03:
not sure how to map those lists to smartctl output at the moment... any ideas?

colttt, 2017-11-03:
hmm not really, but this are all symlinks, so maybe you can make a list and compare it?
```
ls -Alog /dev/disk/by-partlabel/
total 0
lrwxrwxrwx 1 11 Nov  2 12:10 j1d00-ssd -> ../../sdaa1
lrwxrwxrwx 1 11 Nov  2 12:10 j1d01-ssd -> ../../sdab1
lrwxrwxrwx 1 11 Nov  2 12:10 j1d02-ssd -> ../../sdad1
lrwxrwxrwx 1 11 Nov  2 12:10 j1d03-hdd -> ../../sdae1
lrwxrwxrwx 1 11 Nov  2 12:10 j1d04-hdd -> ../../sdaf1
[...]

ls -Alog /dev/disk/by-partlabel/ |awk ' /j/ {print $7 "  " $9}'
j1d00-ssd  ../../sdaa1
j1d01-ssd  ../../sdab1
j1d02-ssd  ../../sdad1
j1d03-hdd  ../../sdae1
j1d04-hdd  ../../sdaf1
[...]
```

colttt, 2017-11-08:
Done in #45

colttt, 2017-11-08:
its done by #46 
the output looks like
```
                {
                        "{#DISKNAME}":"/dev/sdam -d scsi",
                        "{#DISK_PARTNAME}":"j1d17-hdd",
                        "{#DISK_PATH}":"pci-0000:03:00.0-sas-exp0x50030480091018ff-phy33-lun-0-part1",
                        "{#SMART_ENABLED}":"1"
                },
```