v-zhuravlev/zbx-smartctl #48: Zabbix reporting discovery error

Jonmccombs, 2017-11-26:
When using template on Zabbix 3.4 I get the following error under Config - Hosts -Discovery  = Value should be a JSON object. It is a little vague why since I did not alter the script for .pl in any way.

zingreen, 2017-12-01:
I have the same issue on zabbix 3.2.6

v-zhuravlev, 2017-12-02:
please run script by hand and check the output of it

zingreen, 2017-12-02:
my output:

v-zhuravlev, 2017-12-02:
Output is not a valid JSON, you need to suppress those warnings. Try to remove ‘use warnings’ pragma from the script

zingreen, 2017-12-02:
If I run on local console (not ssh) I have output without warnings:  sudo -u zabbix sudo /etc/zabbix/scripts/smartctl-disks-discovery.pl
{
"data":[

{
"{#DISKNAME}":"/dev/sda -d sat",
"{#SMART_ENABLED}":"1"
},
{
"{#DISKNAME}":"/dev/sdb -d sat",
"{#SMART_ENABLED}":"1"
},
{
"{#DISKNAME}":"/dev/sdc -d sat",
"{#SMART_ENABLED}":"1"
},
{
"{#DISKNAME}":"/dev/sdd -d sat",
"{#SMART_ENABLED}":"1"
}
]
}

KaMaToZzz, 2017-12-21:
> Output is not a valid JSON, you need to suppress those warnings. Try to remove ‘use warnings’ pragma from the script

Виталий,
Убрал ‘use warnings’ , но заббикс все равно ругается на: Value should be a JSON object. 
Спасибо.

v-zhuravlev, 2017-12-21:
please try then to use zabbix_get like so:
`zabbix_get -k uHDD.discovery -s <agent_ip>` and then check output with https://jsonlint.com/

KaMaToZzz, 2017-12-21:
Hmm, its look like sudo need's tty?
sudo: no tty present and no askpass program specified

proseti, 2018-02-04:
Hi,
I have the same problem. Added in sudoers /usr/bin/perl didn't resolve the issue :/.

[image]

v-zhuravlev, 2018-05-23:
Check that perl script is exetable:  `chmod +x /etc/zabbix/scripts/smartctl-disks-discovery.pl`

v-zhuravlev, 2018-05-24:
Most of the time, The problem is with permissions. Please tell me if you have solved the issue and what was it so we can update readme with Troubleshooting section.

closed 2018-05-24 by v-zhuravlev

ralga, 2018-06-20:
Hi everyone,
I had this problem when trying to monitor windows. It was not due to permissions but because the JSON sent had a mistake.
The script sent that : 
```
[{
 "data":[
	{
 		"{#DISKSN}":"<SN>",
		"{#DISKMODEL}":"<MODEL>",
		"{#DISKNAME}":"/dev/sda",
		"{#DISKCMD}":"/dev/sda -data",
		"{#SMART_ENABLED}":"1",
		"{#DISKTYPE}":"1"
	},
 ]
}]
```

In this case, there is a comma after the object when there is no other object after. The JSON can't be parse so Zabbix consider it is not a JSON.