v-zhuravlev/zbx-smartctl #49: temperature trigger boundaries

rrauenza, 2017-12-01:
I'm looking at the 2.2 template's triggers and something seems off:

Warning 
{Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uHDD[{#DISKNAME},Temperature_Celsius].avg(1800)}>50& {Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uHDD[{#DISKNAME},Temperature_Celsius].nodata(1800)}=0 

So we get a warning if avg is over 50 

High 
{Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uHDD[{#DISKNAME},Temperature_Celsius].last(0)}>65& {Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uHDD[{#DISKNAME},Temperature_Celsius].last(0)}<100

But we get an error if over 65 _but less than 100_.   

- Why < 100?

v-zhuravlev, 2017-12-02:
to avoid alarms when disk reports invalid data (like 255 in this field). I actually encountered such disks myself. So everything above 100 would be ignored.

closed 2017-12-02 by v-zhuravlev

rrauenza, 2017-12-13:
Thanks -- I supposed if my HD is >= 100 C, it probably isn't responding anymore :)