v-zhuravlev/zbx-smartctl #55: Add support for some HW RAID

b-s-a, 2018-04-13:
Some hardware RAID contollers (like LSI Logic / Symbios Logic SAS2008) creates one standard block device (/dev/sda) and n+1 virtual devices (/dev/sg0 .. /dev/sd2...). /dev/sda is used as block device. It does not supports SMART. One of /dev/sgX (random each time) defined as Virtual Disk and does not support SMART too. But others /dev/sgX translates SMART from real drives.
Current result is:
# /etc/zabbix/scripts/smartctl-disks-discovery.pl
{
	"data":[

		{
			"{#DISKNAME}":"/dev/sda -d scsi",
			"{#SMART_ENABLED}":"0"
		}
	]
}
Expected result (minimum):
# /etc/zabbix/scripts/smartctl-disks-discovery.pl 
{
	"data":[

		{
			"{#DISKNAME}":"/dev/sg0 -d sat",
			"{#SMART_ENABLED}":"1"
		},
		{
			"{#DISKNAME}":"/dev/sg1 -d sat",
			"{#SMART_ENABLED}":"0"
		},
		{
			"{#DISKNAME}":"/dev/sg2 -d sat",
			"{#SMART_ENABLED}":"1"
		}
	]
}
Expected result (preffered):
{
	"data":[

		{
			"{#DISKNAME}":"<model>|<serial>",
			"{#DISKDEV}":"/dev/sg0 -d sat",
			"{#SMART_ENABLED}":"1"
		},
		{
			"{#DISKNAME}":"<model>|<serial>",
			"{#DISKDEV}":"/dev/sg2 -d sat",
			"{#SMART_ENABLED}":"1"
		}
	]
}

v-zhuravlev, 2018-05-23:
Hi. Sorry for late response. Can you please show output of `smartctl --scan-open`

b-s-a, 2018-05-23:
`# smartctl --scan-open`
`/dev/sda -d scsi # /dev/sda, SCSI device`

v-zhuravlev, 2018-05-24:
trry `smartctl --scan`
and `smartctl -i /dev/sg0 -d sat`
would that work?

b-s-a, 2018-05-24:
`# smartctl --scan`
`/dev/sda -d scsi # /dev/sda, SCSI device`

`# smartctl -i /dev/sg0 -d sat`
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.9.0-4-amd64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Toshiba 3.5" MG03ACAxxx(Y) Enterprise HDD
Device Model:     TOSHIBA MG03ACA100
Serial Number:    73QGK41RF
LU WWN Device Id: 5 000039 4ebe828dd
Add. Product Id:  DELL(tm)
Firmware Version: FL1D
User Capacity:    1 000 204 886 016 bytes [1,00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ATA8-ACS (minor revision not indicated)
SATA Version is:  SATA 3.0, 3.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Thu May 24 11:05:41 2018 MSK
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

`# smartctl -i /dev/sg1 -d sat`
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.9.0-4-amd64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: scsi error unsupported scsi opcode

A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.

`# smartctl -i /dev/sg2 -d sat`
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.9.0-4-amd64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Toshiba 3.5" MG03ACAxxx(Y) Enterprise HDD
Device Model:     TOSHIBA MG03ACA100
Serial Number:    73QFK2AEF
LU WWN Device Id: 5 000039 4ebe01055
Add. Product Id:  DELL(tm)
Firmware Version: FL1D
User Capacity:    1 000 204 886 016 bytes [1,00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ATA8-ACS (minor revision not indicated)
SATA Version is:  SATA 3.0, 3.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Thu May 24 11:05:51 2018 MSK
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

v-zhuravlev, 2018-05-24:
ok, now please try:
sudo apt-get install sg3-utils
sudo sg_scan -i
see also https://github.com/v-zhuravlev/zbx-smartctl/pull/29

b-s-a, 2018-05-24:
```
# sg_scan -i
/dev/sg0: scsi0 channel=0 id=0 lun=0
    ATA       TOSHIBA MG03ACA1  FL1D [rmb=0 cmdq=1 pqual=0 pdev=0x0] 
/dev/sg1: scsi0 channel=1 id=0 lun=0
    Dell      Virtual Disk      1028 [rmb=0 cmdq=1 pqual=0 pdev=0x0] 
/dev/sg2: scsi0 channel=0 id=1 lun=0
    ATA       TOSHIBA MG03ACA1  FL1D [rmb=0 cmdq=1 pqual=0 pdev=0x0]
```

v-zhuravlev, 2018-05-24:
please try this script:
https://github.com/v-zhuravlev/zbx-smartctl/commit/d6e5cde1d5ddfe2e9f7b483d4f1a5313979a05ee

b-s-a, 2018-05-24:
```
./smartctl-disks-discovery.pl
{
	"data":[

		{
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 224.
			"{#DISKMODEL}":"",
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 225.
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d scsi",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		}
	]
}
```

b-s-a, 2018-05-24:
I think, after sg_scan should be called smartctl for each drive.

b-s-a, 2018-05-24:
It was ran as root. I forget to copy # symbol.

v-zhuravlev, 2018-05-24:
What is the result then?

b-s-a, 2018-05-24:
I have ran new script once under root user. Output has been posted hour ago.

b-s-a, 2018-05-24:
After commenting `if (-x $sg_scan_cmd){` result is:
```
$ sudo ./smartctl-disks-discovery.pl 
{
	"data":[

		{
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 224.
			"{#DISKMODEL}":"",
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 225.
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d scsi",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		},
		{
			"{#DISKMODEL}":"TOSHIBA MG03ACA100",
			"{#DISKSN}":"73QGK41RF",
			"{#DISKNAME}":"/dev/sg0",
			"{#DISKCMD}":"/dev/sg0 ",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 224.
			"{#DISKMODEL}":"",
Use of uninitialized value in concatenation (.) or string at ./smartctl-disks-discovery.pl line 225.
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/sg1",
			"{#DISKCMD}":"/dev/sg1 ",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		},
		{
			"{#DISKMODEL}":"TOSHIBA MG03ACA100",
			"{#DISKSN}":"73QFK2AEF",
			"{#DISKNAME}":"/dev/sg2",
			"{#DISKCMD}":"/dev/sg2 ",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		}
	]
}
```

v-zhuravlev, 2018-05-24:
try to change -x to -e ?

b-s-a, 2018-05-24:
Same as -x.

b-s-a, 2018-05-24:
I think full path is required

v-zhuravlev, 2018-05-24:
1. strange, it works for me. try to enter full path here: `my $sg_scan_cmd = "sg_scan";`
2. updated script to fix uninitialized values

b-s-a, 2018-05-24:
1. Full path works fine as expected.
2. There are no warnings about uninitialized values.

Is it possible to change device name to something like "$model $sn", because after restart drives will be randomly reassigned?

v-zhuravlev, 2018-05-24:
can I see the output please?

v-zhuravlev, 2018-05-24:
sorry, changing DISKNAME and DISKCMD to serial/model will break too many things.

v-zhuravlev, 2018-05-24:
fixed in https://github.com/v-zhuravlev/zbx-smartctl/commit/106a6f4f6cb992b8591919f93d7edfaaa3f6e2d3

closed 2018-05-24 by v-zhuravlev