v-zhuravlev/zbx-smartctl #56: Wrong discovery by PS1 script

kosfango, 2018-04-15:
Hi All,

Seems, something wrong:

PS C:\Users\Administrator> C:\zabbix\smartmontools\bin\smartctl.exe --scan
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d scsi # /dev/sdb, SCSI device
/dev/csmi0,2 -d ata # /dev/csmi0,2, ATA device
/dev/csmi0,3 -d ata # /dev/csmi0,3, ATA device
/dev/csmi0,4 -d ata # /dev/csmi0,4, ATA device
/dev/csmi0,5 -d ata # /dev/csmi0,5, ATA device

PS C:\Users\Administrator> powershell -NoProfile -ExecutionPolicy Bypass -File C:\zabbix\scripts\smartctl-disks-discovery.ps1
{
 "data":[

        {
                "{#DISKNAME}":"/dev/hda",
                "{#SMART_ENABLED}":"0"
        },

        {
                "{#DISKNAME}":"/dev/hdb",
                "{#SMART_ENABLED}":"0"
        },

        {
                "{#DISKNAME}":"/dev/hdc",
                "{#SMART_ENABLED}":"0"
        }

 ]
}
PS C:\Users\Administrator>

But in the same time this bat file:
@echo off
setlocal enabledelayedexpansion

set smartctl_path=C:\zabbix\smartmontools\bin\smartctl.exe
set serials=dummy_serial
set /a first=1

echo {"data":[
for /f "tokens=1,2,3 delims= " %%i in ('!smartctl_path! --scan') do (
	set smart_enabled=0
    set duplicate=0
    for /F "tokens=3*" %%a in ('!smartctl_path! -i %%i %%j %%k^| find "Serial Number"') do (
        set serial=%%a
    )

        rem duplicate disk entity, check by serial number
        for %%j in (!serials!) do (if %%j == !serial! (set /a duplicate+=1)) 
        set serials=%serials% !serial!

    
	FOR /F "tokens=1" %%a IN ('!smartctl_path! -i %%i %%j %%k ^|find "SMART" ^| find /C "Enabled"') DO (
		if %%a gtr 0 (
        set /a smart_enabled=1
        )
	)
    if not !duplicate! == 1 (
            if not !first! ==1 (
                echo ,
            )
            echo {"{#DISKNAME}":"%%i","{#SMART_ENABLED}":"!smart_enabled!"}
            set /a first=0
    )
)
echo ]}

seems works good:
PS C:\Users\Administrator> C:\zabbix\scripts\smartctl-disks-discovery.bat
{"data":[
{"{#DISKNAME}":"/dev/sda","{#SMART_ENABLED}":"0"}
,
{"{#DISKNAME}":"/dev/sdb","{#SMART_ENABLED}":"0"}
,
{"{#DISKNAME}":"/dev/csmi0,2","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,3","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,4","{#SMART_ENABLED}":"1"}
,
{"{#DISKNAME}":"/dev/csmi0,5","{#SMART_ENABLED}":"1"}
]}
PS C:\Users\Administrator>

lzbohnz, 2018-04-24:
I ran into the same thing. I fixed the PS script, here is the pertinent fragment:

```
$smart_scanresults = & $smartctl "--scan" 

write-host "{"
write-host " `"data`":[`n"
foreach ($smart_scanresult in $smart_scanresults)
{
 
    
    $smartctl_disk_name = $smart_scanresult.Substring(0,$smart_scanresult.IndexOf(" "))
    $smart_enabled = & $smartctl "-i" $smartctl_disk_name | select-string "SMART.+Enabled$"
 
             if($smart_enabled) {            
                $smart_enabled = 1
            } else {            
                $smart_enabled = 0
            }

    
    if ($idx -lt  $smart_scanresults.Count-1)
    {
        $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +"`n`t},`n"
        write-host $line
    }
    elseif ($idx -ge  $smart_scanresults.Count-1)
    {
     
        $line= "`t{`n " + "`t`t`"{#DISKNAME}`":`""+$smartctl_disk_name+"`""+ ",`n" + "`t`t`"{#SMART_ENABLED}`":`""+$smart_enabled+"`"" +"`n`t}"
        write-host $line
    }
    $idx++;
}
```

This gives me the appropriate /dev/sda, /dev/hda, etc. rather than having it hard-coded as hda.

kosfango, 2018-04-24:
Btw, in my case it works too:

https://github.com/v-zhuravlev/zbx-smartctl/pull/52

v-zhuravlev, 2018-05-23:
Thanks @lzbohnz ,

commit fe3576c9 (2018-05-23): fixes #56, thanks @lzbohnz
closed 2018-05-23 by v-zhuravlev