v-zhuravlev/zbx-smartctl #61: Discovery rules Value should be a JSON object

roderik-k, 2018-06-25:
The discovery script ps1 for Win has a small mistake, its "comma" on line 20
in this case is not valid Json format, I dont know to correct the ps1 script to reduce  ","
[image]

thanks

zabbix_get:
--------------
{
 "data":[

	{
 		"{#DISKSN}":"xxxxxxxxxx",
		"{#DISKMODEL}":"SanDisk SD8SBAT-256G-1006",
		"{#DISKNAME}":"/dev/sda",
		"{#DISKCMD}":"/dev/sda -data",
		"{#SMART_ENABLED}":"1",
		"{#DISKTYPE}":"1"
	},

	{
 		"{#DISKSN}":"yyyyyyyyyyy",
		"{#DISKMODEL}":"hp PLDS DVDRW  DU8AESH",
		"{#DISKNAME}":"/dev/csmi0,1",
		"{#DISKCMD}":"/dev/csmi0,1 -data",
		"{#SMART_ENABLED}":"0",
		"{#DISKTYPE}":"2"
	},

 ]

v-zhuravlev, 2018-06-27:
thanks, will try to fix it ASAP

roderik-k, 2018-06-27:
thx for your support, 👍

SimplyKot, 2018-06-27:
The same error:
`{
 "data":[

        {
                "{#DISKSN}":"Y21WK8JC",
                "{#DISKMODEL}":"Maxtor 6Y080M0",
                "{#DISKNAME}":"/dev/sda",
                "{#DISKCMD}":"/dev/sda -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
        },

        {
                "{#DISKSN}":"",
                "{#DISKMODEL}":"",
                "{#DISKNAME}":"/dev/sdb",
                "{#DISKCMD}":"/dev/sdb -dscsi",
                "{#SMART_ENABLED}":"0",
                "{#DISKTYPE}":"2"
        },

        {
                "{#DISKSN}":"9QM7W1LD",
                "{#DISKMODEL}":"ST3500320AS",
                "{#DISKNAME}":"/dev/csmi0,1",
                "{#DISKCMD}":"/dev/csmi0,1 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
        },

        {
                "{#DISKSN}":"Z2AMFRRB",
                "{#DISKMODEL}":"ST500DM002-1BD142",
                "{#DISKNAME}":"/dev/csmi0,2",
                "{#DISKCMD}":"/dev/csmi0,2 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
        },

        {
                "{#DISKSN}":"Z3T83S3M",
                "{#DISKMODEL}":"ST500DM002-1BD142",
                "{#DISKNAME}":"/dev/csmi0,3",
                "{#DISKCMD}":"/dev/csmi0,3 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
        },

 ]
}
`

v-zhuravlev, 2018-07-05:
mykolq  what powershell and Windowns versions do you have?

mykolq, 2018-07-05:
5.1.14393.2248

mykolq, 2018-07-05:
sure. ```
C:\zabbix_agent\smartmontools\bin>smartctl --scan-open
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/csmi2,0 -d ata # /dev/csmi2,0, ATA device
/dev/csmi2,1 -d ata # /dev/csmi2,1, ATA device
```

v-zhuravlev, 2018-07-05:
Strange, can't find the reason of this extra comma.  Fixed problem with trim() function though(I hope): https://github.com/v-zhuravlev/zbx-smartctl/commit/9177c28b3dc26c793bb6283b1213eb5fffa5b076

v-zhuravlev, 2018-07-05:
Ok, found the problem please try the latest commit

mykolq, 2018-07-05:
thanks a lot =)  it works fine

roderik-k, 2018-07-05:
thanks it works very nice, valid json format

roderik-k, 2018-07-06:
I have this one host (dell) where something not working correctly: 
C:\Program Files\smartmontools\bin>smartctl.exe --scan-open
/dev/sda -d ata # /dev/sda, ATA device
/dev/csmi0,0 -d ata # /dev/csmi0,0, ATA device

valid json - powershell cmd on dell host local:
PS C:\zabbix_agent> .\smartctl-disks-discovery.ps1
{
 "data":[

        {
                "{#DISKSN}":"171616DFD067",
                "{#DISKMODEL}":"Micron 1100 SATA 256GB",
                "{#DISKNAME}":"/dev/sda",
                "{#DISKCMD}":"/dev/sda -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"2"
        }

 ]
}
=======================================
but in zabbix system is not valid json ("comma" error):
[root@monitoring ~]# zabbix_get -s 192.168.1.221 -p 10050 -k uHDD.discovery
{
 "data":[

	{
 		"{#DISKSN}":"171616DFD067",
		"{#DISKMODEL}":"Micron 1100 SATA 256GB",
		"{#DISKNAME}":"/dev/sda",
		"{#DISKCMD}":"/dev/sda -data",
		"{#SMART_ENABLED}":"1",
		"{#DISKTYPE}":"1"
	},

 ]
}

v-zhuravlev, 2018-07-07:
Sorry! Was trying to fix that in a rush. Now it should work in your case as well.

SimplyKot, 2018-07-09:
Still not working...
`Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Невозможно найти тип [ordered]: убедитесь в том, что сборка, содержащая этот тип, загружена.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:93 знак:24
+     $disks += [ordered] <<<< @{
    + CategoryInfo          : InvalidOperation: (ordered:String) [], RuntimeException
    + FullyQualifiedErrorId : TypeNotFound

Имя "ConvertTo-Json" не распознано как имя командлета, функции, файла скрипта или выполняемой программы. Проверьте правильность написания имени, а также наличие и правильность пути, после чего повторите попытку.
C:\zabbix\conf\scripts\smartctl-disks-discovery.ps1:104 знак:46
+ $json_disks = @{data= $disks}| ConvertTo-Json <<<<  -Depth 2
    + CategoryInfo          : ObjectNotFound: (ConvertTo-Json:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException`

v-zhuravlev, 2018-07-09:
What windows and powershell version do you have?

v-zhuravlev, 2018-07-09:
Ok, no fancy PS modules dependencies now, just fixed comma thing :)

closed 2018-07-09 by v-zhuravlev

SimplyKot, 2018-07-10:
Now working great! Thank you!

v-zhuravlev, 2018-07-10:
Finally ! 😆

zetheroo, 2019-05-09:
I am experiencing this issue on a Debian Stretch (PVE) system running zabbix_agent 4.0.7

```
root@proxsat:/etc/zabbix/scripts# ./smartctl-disks-discovery.pl 
{
	"data":[

		{
			"{#DISKMODEL}":"WDC WD1003FZEX-00K3CA0",
			"{#DISKSN}":"WD-WCC6Y0NYF3E7",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"WDC WD1003FZEX-00K3CA0",
			"{#DISKSN}":"WD-WCC6Y4KD46L2",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"Samsung SSD 860 PRO 1TB",
			"{#DISKSN}":"S42NNF0K501976K",
			"{#DISKNAME}":"/dev/sdc",
			"{#DISKCMD}":"/dev/sdc -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"Samsung SSD 860 PRO 1TB",
			"{#DISKSN}":"S42NNF0K501910F",
			"{#DISKNAME}":"/dev/sdd",
			"{#DISKCMD}":"/dev/sdd -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}

```

TiratoreA, 2019-07-03:
[image]

I have the same error in Linux

TiratoreA, 2019-07-03:
# smartctl --scan-open
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d sat # /dev/sdb [SAT], ATA device
/dev/bus/0 -d megaraid,0 # /dev/bus/0 [megaraid_disk_00], SCSI device
/dev/bus/0 -d megaraid,1 # /dev/bus/0 [megaraid_disk_01], SCSI device
/dev/bus/0 -d megaraid,2 # /dev/bus/0 [megaraid_disk_02], SCSI device
/dev/bus/0 -d megaraid,3 # /dev/bus/0 [megaraid_disk_03], SCSI device
/dev/bus/0 -d megaraid,4 # /dev/bus/0 [megaraid_disk_04], SCSI device
/dev/bus/0 -d megaraid,5 # /dev/bus/0 [megaraid_disk_05], SCSI device
/dev/bus/0 -d megaraid,6 # /dev/bus/0 [megaraid_disk_06], SCSI device
/dev/bus/0 -d megaraid,7 # /dev/bus/0 [megaraid_disk_07], SCSI device

commit 81f280fe (2018-07-09): Merge branch 'fix_ps_json'
- PS script: reverted back to using write-host, fixes #61