v-zhuravlev/zbx-smartctl #62: Braking uHDD parameter if using comma in #DISKNAME or #DISKCMD

SimplyKot, 2018-07-12:
Don't working with this configuration:
`{
 "data":[
         {
                "{#DISKSN}":"9QM7W1LD",
                "{#DISKMODEL}":"ST3500320AS",
                "{#DISKNAME}":"/dev/csmi0,1",
                "{#DISKCMD}":"/dev/csmi0,1 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
        
         }
 ]
}`

Example:
`uHDD["/dev/csmi0,1 -data",Temperature_Celsius]`
execute following command on the host:
`for /F "tokens=10 usebackq" %a in (`""C:\Program Files\smartmontools\bin\smartctl.exe" -A /dev/csmi0 | find "1""`) do @echo %a`

v-zhuravlev, 2018-07-12:
No idea how to fix it now... how about  `uHDD.A[*]` ?

SimplyKot, 2018-07-12:
> how about uHDD.A[*] ?

 It's working. Thank you.

closed 2018-07-12 by SimplyKot