v-zhuravlev/zbx-smartctl #67: smartctl-disks-discovery.pl - not running properly

elbrinke, 2018-08-28:
Hi, 

When i try to run the "smartctl-disks-discovery.pl" script as zabbix it gives the following error:

```
sudo -u zabbix ./smartctl-disks-discovery.pl 
sg_scan: Error opening /dev/sg0 : Permission denied
sg_scan: Error opening /dev/sg1 : Permission denied
sg_scan: Error opening /dev/sg2 : Permission denied
Smartctl open device: /dev/sda failed: Permission denied
Smartctl open device: /dev/sdb failed: Permission denied
Smartctl open device: /dev/sdc failed: Permission denied
```
I have added the sudoers rule: 

`zabbix ALL=(ALL) NOPASSWD: /usr/sbin/smartctl,/etc/zabbix/scripts/smartctl-disks-discovery.pl`

But i still get the same issue. 

Can you please assist? 
Regards
E.

v-zhuravlev, 2018-08-28:
`su - zabbix`
`./smartctl-disks-discovery.pl`
?

elbrinke, 2018-08-29:
What permissions should this `smartctl-disks-discovery.pl` script have? It is currently owned by root.

elbrinke, 2018-08-29:
`su - zabbix /etc/zabbix/scripts/smartctl-disks-discovery.pl`

asked for zabbix password?

v-zhuravlev, 2018-08-29:
zabbix user must be able to run it

elbrinke, 2018-08-29:
```
s -l /etc/zabbix/scripts/
total 8
-rwxrwxrwx 1 zabbix zabbix 7201 Aug 28 14:12 smartctl-disks-discovery.pl
```

I changed the script permissions to zabbix as well, but nothing works.

elbrinke, 2018-08-29:
Should the zabbix user be part of a specif group? Below is the sudoers file:

```
Defaults:zabbix !requiretty
zabbix ALL=(ALL) NOPASSWD: /usr/sbin/smartctl,/etc/zabbix/scripts/smartctl-disks-discovery.pl
```

v-zhuravlev, 2018-08-29:
no.

elbrinke, 2018-08-29:
So what must the permissions be? I followed exactly what you posted.

v-zhuravlev, 2018-08-29:
root:root rwxr-x-r-x

elbrinke, 2018-08-29:
total 8
-rwxr-xr-x 1 root root 7201 Aug 28 14:12 smartctl-disks-discovery.pl

elbrinke, 2018-08-29:
Should this: `Defaults:zabbix !requiretty` also be added into the sudoers file?

elbrinke, 2018-08-29:
```
sudo su - zabbix
No directory, logging in with HOME=/
This account is currently not available.
```
But i can see the account... 

```
id zabbix 
uid=114(zabbix) gid=122(zabbix) groups=122(zabbix)
```

v-zhuravlev, 2018-08-29:
https://github.com/v-zhuravlev/zbx-smartctl/pull/35 !requiretty was added to fix some specific issue. Generally, this is not required.

v-zhuravlev, 2018-08-29:
you can try to use `zabbix_get` utility with discovery key to test it how zabbix sees it then

elbrinke, 2018-08-29:
On Zabbix Server:

**zabbix_get**

```
Command 'zabbix_get' not found, but can be installed with:

apt install zabbix-proxy-mysql  
apt install zabbix-proxy-pgsql  
apt install zabbix-proxy-sqlite3
apt install zabbix-server-mysql 
apt install zabbix-server-pgsql 
```

elbrinke, 2018-08-29:
Installed **zabbix-get** package from Ubuntu repo. 

```
apt install zabbix-get 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  zabbix-get
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 62.1 kB of archives.
After this operation, 215 kB of additional disk space will be used.
Get:1 http://repo.zabbix.com/zabbix/3.4/ubuntu bionic/main amd64 zabbix-get amd64 1:3.4.13-1+bionic [62.1 kB]
Fetched 62.1 kB in 1s (43.8 kB/s)     
Selecting previously unselected package zabbix-get.
(Reading database ... 31639 files and directories currently installed.)
Preparing to unpack .../zabbix-get_1%3a3.4.13-1+bionic_amd64.deb ...
Unpacking zabbix-get (1:3.4.13-1+bionic) ...
Processing triggers for man-db (2.8.3-2) ...
Setting up zabbix-get (1:3.4.13-1+bionic) ...

```

elbrinke, 2018-08-29:
```
zabbix_get -s127.0.01 -p10050 -kagent.version
3.4.13
```

v-zhuravlev, 2018-09-13:
any luck?

elbrinke, 2018-09-19:
Hi @v-zhuravlev , It is still a problem:
```
● zabbix-agent.service - Zabbix Agent
   Loaded: loaded (/lib/systemd/system/zabbix-agent.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-09-19 16:31:30 SAST; 2s ago
  Process: 13031 ExecStart=/usr/sbin/zabbix_agentd -c $CONFFILE (code=exited, status=0/SUCCESS)
 Main PID: 13033 (zabbix_agentd)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/zabbix-agent.service
           ├─13033 /usr/sbin/zabbix_agentd -c /etc/zabbix/zabbix_agentd.conf
           ├─13057 /usr/sbin/zabbix_agentd: collector [idle 1 sec]
           ├─13076 /usr/sbin/zabbix_agentd: listener #1 [waiting for connection]
           ├─13087 /usr/sbin/zabbix_agentd: listener #2 [waiting for connection]
           ├─13096 /usr/sbin/zabbix_agentd: listener #3 [waiting for connection]
           └─13113 /usr/sbin/zabbix_agentd: active checks #1 [idle 1 sec]

Sep 19 16:31:30 gandalf systemd[1]: Starting Zabbix Agent...
Sep 19 16:31:30 gandalf systemd[1]: zabbix-agent.service: Can't open PID file /run/zabbix/zabbix_agentd.pid (yet?) after start: No such file or directory
Sep 19 16:31:30 gandalf systemd[1]: zabbix-agent.service: Supervising process 13033 which is not our child. We'll most likely not notice when it exits.
Sep 19 16:31:30 gandalf systemd[1]: Started Zabbix Agent.
```
I still cannot read the smartmon info.

elbrinke, 2018-09-19:
Hi @v-zhuravlev , I did the following checks:
```

root@zabbix:~# zabbix_get -s 192.168.1.6 -p 10050 -k "uHDD.discovery"
ZBX_NOTSUPPORTED: Timeout while executing a shell script.
```

bkar07, 2018-12-29:
> root@zabbix:~# zabbix_get -s 192.168.1.6 -p 10050 -k "uHDD.discovery"
> ZBX_NOTSUPPORTED: Timeout while executing a shell script.

Hi,
I solved it by setting Timeout=8 in zabbix_agentd.conf.

nerijus, 2019-01-27:
Timeout=8 in zabbix_agentd.conf helped for me too.

closed 2019-04-19 by v-zhuravlev

jiansong, 2019-11-26:
Information for anyone runs again into the same problem: 
All of above tweaks are not necessary, because you invoked wrong command to test it. 

Wrong:
`$ sudo -u zabbix /etc/zabbix/scripts/smartctl-disks-discovery.pl`

Corret:
`$ sudo -u zabbix sudo /etc/zabbix/scripts/smartctl-disks-discovery.pl`