v-zhuravlev/zbx-smartctl #70: Zabbix 4.0 Template Import

XplizitBe, 2018-10-22:
I have a problem with importing the template into Zabbix 4.0. It imports without errors but only application, and discovery script get import. No items no triggers?

Querying by hand through zabbix_get works fine.

mykolq, 2018-10-25:
it seems there is a problem with regexp in preprocessing. you can fix it by yourself by 
taking the expression in parentheses and getting non first matching. example:
[image]

v-zhuravlev, 2018-11-13:
@XplizitBe  which template? Template for 3.4, for example, doesn't have any items or triggers - everything is discovered by LLD, so see the discovery section inside the template - where you would find items and trigger prototypes.

mykolq, 2018-12-10:
[image]
something like that.  You can check expressions here [https://regex101.com/](url).

v-zhuravlev, 2019-04-19:
@jpaulickcz , can you please attach `smartctl -A` output?

v-zhuravlev, 2019-04-24:
closing this now, please reopen if still having issues.

closed 2019-04-24 by v-zhuravlev