v-zhuravlev/zbx-smartctl #71: smartctl-disks-discovery.pl: command not found Mac os X

roderik-k, 2018-11-20:
Dear Vitaly,
I would like to use smartctl on my mac os x,
-installed latest smartmontools utility
-Installed the script smartctl-disks-discovery.pl in /etc/zabbix/scripts/
-added the following permissions into /etc/sudoers

zabbix server - discovery info: Value should be a JSON object.
zabbix_get -s 192.168.1.241 -p 10050 -k uHDD.discovery
results are: empty line
I dont know how to correct error "command not found" (log file)

zabbix_agent.log
712:20181120:092903.201 Starting Zabbix Agent [MacBookProH]. Zabbix 2.2.5 (revision 47411).
   712:20181120:092903.202 using configuration file: /usr/local/etc/zabbix/zabbix_agentd.conf
   713:20181120:092903.204 agent #0 started [collector]
   714:20181120:092903.206 agent #1 started [listener #1]
   715:20181120:092903.207 agent #2 started [listener #2]
   717:20181120:092903.208 agent #4 started [active checks #1]
   716:20181120:092903.209 agent #3 started [listener #3]
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found
sudo: /etc/zabbix/scripts/smartctl-disks-discovery.pl: command not found

Thanks for your support

roderik-k, 2018-11-21:
My experiencies about this problem:
1) smartctl tool path is /usr/local/sbin/
I changed the path in sudoers
2) smartctl-disks-discovery.pl: command not found, stpl scrip doesnt have permissions to execute, my solution: sudo cmod +x /etc/zabbix/scripts/smartctl-disks-discovery.pl
3) now resutls are: sudo /etc/zabbix/scripts/smartctl-disks-discovery.pl 
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"",
			"{#DISKNAME}":"disk0",
			"{#DISKCMD}":"disk0",
			"{#SMART_ENABLED}":"0",
Use of uninitialized value in concatenation (.) or string at /etc/zabbix/scripts/smartctl-disks-discovery.pl line 233.
			"{#DISKTYPE}":""
		}
	]
}

4)zabbix_get -s 192.168.1.241 -p 10050 -k uHDD.discovery
empty line

v-zhuravlev, 2018-11-21:
can you run `smartctl` without the path? try to adjust variable in the script here:
https://github.com/v-zhuravlev/zbx-smartctl/blob/master/discovery-scripts/nix/smartctl-disks-discovery.pl#L9

roderik-k, 2018-11-21:
added path to smartctl, export PATH=$PATH:/usr/local/sbin
smartctl working in terminal without path
MacBookProH:~ RK$ smartctl --scan-open
IOService:/AppleACPIPlatformExpert/PCI0/AppleACPIPCI/SATA@1F,2/AppleICH8AHCI/PRT0@0/IOAHCIDevice@0/AppleAHCIDiskDriver/IOAHCIBlockStorageDevice -d ata # IOService:/AppleACPIPlatformExpert/PCI0/AppleACPIPCI/SATA@1F,2/AppleICH8AHCI/PRT0@0/IOAHCIDevice@0/AppleAHCIDiskDriver/IOAHCIBlockStorageDevice, ATA device

new script results:
sudo /etc/zabbix/scripts/smartctl-disks-discovery.pl 
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 12: use: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 13: use: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 16: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 19: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 20: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 21: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 22: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 23: my: command not found
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 25: syntax error near unexpected token `{'
/etc/zabbix/scripts/smartctl-disks-discovery.pl: line 25: `if ( $^O eq 'darwin' ) {    # if MAC OSX (limited support, consider to use smartctl --scan-open)'

v-zhuravlev, 2019-04-19:
It seems that something is wrong with the script you are using. Have you tried to reinstall it?

v-zhuravlev, 2019-04-24:
Closing it now, Please try with the new version of template and scripts. Reopen, if still having issues.

closed 2019-04-24 by v-zhuravlev