v-zhuravlev/zbx-smartctl #74: No devices were found in version 4.0.2

yukunyi, 2018-12-20:
I imported the 3.4 template, set up the script and permissions, the script execution results are correct, but no devices have been found in zabbix. Does it not support the 4.x version?

c0r3dump3d, 2019-02-01:
I have the same issue ... the execution of script for disk autodiscovery is correct. I'm  also using Zabbix 4.0.2 and I have no erros in logs ...

c0r3dump3d, 2019-02-08:
I have resolved the issue. It's cause for the timeout parameter in zabbix-agent configuration, in my system I have many disk and the perl script takes more time that the timeout value. For example you can check  for the HDD  autodiscovery with:

``
   zabbix_agentd -t uHDD.discovery
``

If you can see the message:

``uHDD.discovery                                [m|ZBX_NOTSUPPORTED] [Timeout while executing a shell script.]
``

You need to change in `/etc/zabbix/zabbix_agentd.conf` the value of the option timeout, for example to max value:

`Timeout=30`

yukunyi, 2019-02-09:
@c0r3dump3d This is the problem in this place, I also encountered another problem on the centos system, the sudo command was intercepted by selinux, I just closed selinux. Now my two systems here are recognized correctly. Thank you!

closed 2019-04-19 by v-zhuravlev