v-zhuravlev/zbx-smartctl #79: Zabbix 3.0 - Discovery rules problem

Snowmanko, 2019-02-26:
Hi,

I am experiencing error `Cannot create item: Incorrect function 'last' expression. Check expression part starting from "uHDD[".` in Discovery rule - Template_HDD_SMARTMONTOOLS_2_WITH_LLD: smartctl-disks-discovery, can you advice what could cause this problem ?

I suppose there is some problem with .last expressions in Triggers.

Thanks for any advice.
BR!

v-zhuravlev, 2019-04-19:
Hi!
Which version of template do you use?
What name of the disk that caused the error?

v-zhuravlev, 2019-04-24:
Please reopen if still having this problem.

Snowmanko, 2019-04-24:
Sorry, I didn't see that you replied. I'm running Zabbix 3.0.25.
I can see only -> Cannot create item: Incorrect function 'last' expression. Check expression part starting from: "uHDD[".
.. in hosts "discovery rules" on name "Template_HDD_SMARTMONTOOLS_2_WITH_LLD: smartctl-disks-discovery".

How can I see full error, its probably cut because of the length by zabbix.

Snowmanko, 2019-04-24:
Only one with Type calculated is : {#DISKNAME}: SMART critical errors total
Formula is : 
`last("uHDD["{#DISKCMD}",5 Reallocated]")+last("uHDD["{#DISKCMD}",Spin_Retry_Count]")+last("uHDD["{#DISKCMD}",Current_Pending_Sector]")+last("uHDD["{#DISKCMD}",Offline_Uncorrectable]")`

Attaching exported XML, has to be zipped bcs of guthub.
[zbx_export_templates.xml.zip](https://github.com/v-zhuravlev/zbx-smartctl/files/3113136/zbx_export_templates.xml.zip)

Snowmanko, 2019-04-24:
Error looks like this
[image]

v-zhuravlev, 2019-04-24:
Do you have any items actually created?

v-zhuravlev, 2019-04-24:
Can you run discovery script manually and show me the output?

Snowmanko, 2019-05-02:
Sure,

`{
	"data":[

		{
			"{#DISKMODEL}":"INTEL SSD",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"INTEL SSD",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"ST",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdc",
			"{#DISKCMD}":"/dev/sdc -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"ST",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdd",
			"{#DISKCMD}":"/dev/sdd -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"ST",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sde",
			"{#DISKCMD}":"/dev/sde -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"ST",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdf",
			"{#DISKCMD}":"/dev/sdf -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"ST",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdg",
			"{#DISKCMD}":"/dev/sdg -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"INTEL SSD",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdh",
			"{#DISKCMD}":"/dev/sdh -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"INTEL SSD",
			"{#DISKSN}":"XY",
			"{#DISKNAME}":"/dev/sdi",
			"{#DISKCMD}":"/dev/sdi -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}
`

Snowmanko, 2019-05-02:
> Do you have any items actually created?
[image]

Yes, this is from template. There are also correct items.