v-zhuravlev/zbx-smartctl #81: Не корректные данные параметров для Smartctl

raegent, 2019-04-12:
Добрый день, спасибо за вашу работу. 
Имею Zabbix 4.2, Ubuntu 18.04.2, smartctl 6.5 2016-01-24 r4214 [x86_64-linux-4.15.0-47-generic]

Вывод - все отлично: /etc/zabbix/scripts/smartctl-disks-discovery.pl
{
        "data":[

                {
                        "{#DISKMODEL}":"64GB SSD",
                        "{#DISKSN}":"H42795J001891",
                        "{#DISKNAME}":"/dev/sda",
                        "{#DISKCMD}":"/dev/sda -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                },
                {
                        "{#DISKMODEL}":"WDC WD40PURZ-85TTDY0",
                        "{#DISKSN}":"WD-WCC7K0TRFN79",
                        "{#DISKNAME}":"/dev/sdb",
                        "{#DISKCMD}":"/dev/sdb -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"0"
                },

Но в самом мониторинге имею текст ошибки: 
smartctl 6.5 2016-01-24 r4214 [x86_64-linux-4.15.0-47-generic] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Smartctl open device: /dev//dev/sda [SAT] failed: No such device

Лишний /dev/ как я полагая. Где это я смогу поправить?

v-zhuravlev, 2019-04-12:
а какой шаблон используете?

raegent, 2019-04-12:
> 
> 
> а какой шаблон используете?

Template_3.4_HDD_SMARTMONTOOLS_2_WITH_LLD.xml

v-zhuravlev, 2019-04-12:
Возможно кто-то в шаблон вносил изменения? Если посмотреть файл https://github.com/v-zhuravlev/zbx-smartctl/blob/master/Template_3.4_HDD_SMARTMONTOOLS_2_WITH_LLD.xml то тут нигде dev не упоминается, поэтому задвоения быть не должно.

v-zhuravlev, 2019-04-19:
Проблема все еще актуальна?

v-zhuravlev, 2019-04-24:
Откройте заново, еесли еще актуально. Рекомендую также обновить шаблон, userparameters, скрипт, сегодня внес большие обновления.

closed 2019-04-24 by v-zhuravlev