v-zhuravlev/zbx-smartctl #85: NVMe support ?

brosky, 2019-04-22:
Hi have a machine with two NVMEs in raid1 configuration and: 

```
[root@scripts]# ./smartctl-disks-discovery.pl
{
	"data":[

	]
}

```
Other info:

```
[root@scripts]# smartctl -i /dev/nvme0n1
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.5.1.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SAMSUNG MZVLB512HAJQ-00000
Serial Number:                      S3W8NX0M108020
Firmware Version:                   EXA7301Q
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 512,110,190,592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Utilization:            18,869,055,488 [18.8 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Mon Apr 22 13:44:08 2019 CEST

```

```
[root@scripts]# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     S3W8NX0M108020       SAMSUNG MZVLB512HAJQ-00000               1          18.87  GB / 512.11  GB    512   B +  0 B   EXA7301Q
/dev/nvme1n1     S3W8NX0M150165       SAMSUNG MZVLB512HAJQ-00000               1         511.77  GB / 512.11  GB    512   B +  0 B   EXA7301Q
```

Is there something that I can do so that the disks are visible ?

v-zhuravlev, 2019-04-22:
please run 
`smartctl -i -H -A  /dev/nvme0n1`
`smartctl --scan-open`

brosky, 2019-04-22:
Hi, 

```
~]# smartctl -i -H -A /dev/nvme0n1
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.5.1.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SAMSUNG MZVLB512HAJQ-00000
Serial Number:                      S3W8NX0M108020
Firmware Version:                   EXA7301Q
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 512,110,190,592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Utilization:            18,870,272,000 [18.8 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Mon Apr 22 14:40:43 2019 CEST

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02, NSID 0x1)
Critical Warning:                   0x00
Temperature:                        30 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    5,450,838 [2.79 TB]
Data Units Written:                 172,552 [88.3 GB]
Host Read Commands:                 469,749,169
Host Write Commands:                14,437,203
Controller Busy Time:               160
Power Cycles:                       6
Power On Hours:                     153
Unsafe Shutdowns:                   2
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               30 Celsius
Temperature Sensor 2:               33 Celsius
```

```
[root@dcos]# smartctl --scan-open
# scan_smart_devices: glob(3) aborted matching pattern /dev/discs/disc*
```

v-zhuravlev, 2019-04-23:
what is nvme cli tool?

brosky, 2019-04-23:
Hi Vitaly, 

[nvme-cli ](https://github.com/linux-nvme/nvme-cli )is a tool like smartctl for NVME's, comes by default in centos7:

Name        : nvme-cli
Arch        : x86_64
Version     : 1.6
Release     : 4.el7_6
Size        : 251 k
Repo        : updates/7/x86_64
Summary     : NVMe management command line interface
URL         : https://github.com/linux-nvme/nvme-cli
License     : GPLv2+
Description : nvme-cli provides NVM-Express user space tooling for **Linux.**

Anyways, I think the problem is that smartctl-disks-discovery.pl looks in /dev/discs/disc* while NVME's are connected directly to the PCI bus - therefore their device link is in /dev/nvme*

v-zhuravlev, 2019-04-23:
`nvme list` should be added to discovery script then, just like `sg-scan`. Then we can append nvme output to the one from smartctl --scan-open. Template should be uploaded as well to parse different output.

v-zhuravlev, 2019-04-23:
1. Here is the modified discovery script:
https://github.com/v-zhuravlev/zbx-smartctl/commit/aff95cc2714eeb993bdb5f88f2f6c56bd6ee63b2
2. Here you can find modified template for 3.4 + new instructions how to setup the template (requires to update userparameters): 
https://github.com/v-zhuravlev/zbx-smartctl/tree/nvme

brosky, 2019-04-23:
discovery now works nice, great job !

```
[root@ scripts]# ./smartctl-disks-discovery.pl
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"S3W8NX0M108020",
			"{#DISKNAME}":"/dev/nvme0n1",
			"{#DISKCMD}":"/dev/nvme0n1",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		},
		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"S3W8NX0M150165",
			"{#DISKNAME}":"/dev/nvme1n1",
			"{#DISKCMD}":"/dev/nvme1n1",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		}
	]
}
```

Unfortunately I have zabbix 3.2.11 and the template does not import: 

```
Import failed
Details
Invalid tag "/zabbix_export/version": unsupported version number.
```

Is it possible to create a template for 3.2 ?

v-zhuravlev, 2019-04-23:
I see few issues with the discovery output, let me fix that...

v-zhuravlev, 2019-04-23:
As for template, unfortunately I don't have time to support 3.2 template. Sorry. But you can try to port a template for 3.2. 
Note though instead of regex preprocessing used in 3.4, preprocessing should be in userparameters on agent side in 3.2.
I'll try to post ATA vs SAS vs NVMe metric matrix soon.

v-zhuravlev, 2019-04-23:
ok updated script in 77c4a725f84204a4e80efd1712af33f7f50ae0bf . please check if output is with {#DISKTYPE} = 1 now.

v-zhuravlev, 2019-04-24:
Would you be able to provide output of the new script? Thanks.

brosky, 2019-04-24:
Sorry for the delay: 

scripts]# ./smartctl-disks-discovery.pl
{
	"data":[

		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M108020",
			"{#DISKNAME}":"/dev/nvme0n1",
			"{#DISKCMD}":"/dev/nvme0n1",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M150165",
			"{#DISKNAME}":"/dev/nvme1n1",
			"{#DISKCMD}":"/dev/nvme1n1",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"1"
		}
	]
}

v-zhuravlev, 2019-04-24:
great, now output is as expected (DISKTYPE = 1, and DISKMODEL filled). Thanks!

v-zhuravlev, 2019-04-24:
can you please also run `smartctl -H -i -A -l background -l error nvme0n1` ?

brosky, 2019-04-24:
```
scripts]# smartctl -H -i -A -l background -l error nvme0n1
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.5.1.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Smartctl open device: nvme0n1 failed: No such device
```

if I put the full path then it works: 

```
scripts]# smartctl -H -i -A -l background -l error /dev/nvme0n1
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.5.1.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SAMSUNG MZVLB512HAJQ-00000
Serial Number:                      S3W8NX0M108020
Firmware Version:                   EXA7301Q
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 512,110,190,592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Utilization:            19,033,997,312 [19.0 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Wed Apr 24 13:21:07 2019 CEST

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02, NSID 0x1)
Critical Warning:                   0x00
Temperature:                        30 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    5,450,840 [2.79 TB]
Data Units Written:                 186,554 [95.5 GB]
Host Read Commands:                 469,749,177
Host Write Commands:                15,874,972
Controller Busy Time:               169
Power Cycles:                       6
Power On Hours:                     168
Unsafe Shutdowns:                   2
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               30 Celsius
Temperature Sensor 2:               33 Celsius

Error Information (NVMe Log 0x01, max 64 entries)
No Errors Logged
```

v-zhuravlev, 2019-04-24:
@brosky , one more thing, can you also run `smartctl --scan-open -d nvme` ? Thank you

brosky, 2019-04-25:
Sure:

```
]# smartctl --scan-open -d nvme
/dev/nvme0 -d nvme # /dev/nvme0, NVMe device
/dev/nvme1 -d nvme # /dev/nvme1, NVMe device
```

v-zhuravlev, 2019-04-25:
@brosky , sorry to bother you again, can you please try new version of the discovery script?
https://github.com/v-zhuravlev/zbx-smartctl/blob/nvme_discovery/discovery-scripts/nix/smartctl-disks-discovery.pl
It should discovery disks even without nvme-cli if I got it right. Also it should mark properly NVMe devices with {#SMART_ENABLED} = 1

brosky, 2019-04-25:
With nvme installed:

[root@scripts]# ./smartctl-disks-discovery.pl2
```
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/discs/disc*",
			"{#DISKCMD}":"/dev/discs/disc* -d nvme",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M108020",
			"{#DISKNAME}":"/dev/nvme0n1",
			"{#DISKCMD}":"/dev/nvme0n1",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M150165",
			"{#DISKNAME}":"/dev/nvme1n1",
			"{#DISKCMD}":"/dev/nvme1n1",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}
```

Removed nvme:

```
[root@scripts]# rpm -qa|grep nvme
nvme-cli-1.6-4.el7_6.x86_64
[root@scripts]# rpm -e nvme-cli
```
Run again the script:

```
[root@scripts]# ./smartctl-disks-discovery.pl2
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/discs/disc*",
			"{#DISKCMD}":"/dev/discs/disc* -d nvme",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}
```

So without nvme-cli it has a incomplete result.

v-zhuravlev, 2019-04-25:
`smartctl -x /dev/discs/disc* -d nvme` ? :)

brosky, 2019-05-06:
Sorry for delay, vacation.

```
[root@ scripts]# smartctl -x /dev/discs/disc* -d nvme
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.5.1.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

Smartctl open device: /dev/discs/disc* failed: No such device
```

nerijus, 2019-05-16:
> @brosky , sorry to bother you again, can you please try new version of the discovery script? [...]

I tried it, and with nvme disks its output is:
```
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/discs/disc*",
			"{#DISKCMD}":"/dev/discs/disc* -d nvme",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M418376",
			"{#DISKNAME}":"/dev/nvme0n1",
			"{#DISKCMD}":"/dev/nvme0n1",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
			"{#DISKSN}":"S3W8NX0M418304",
			"{#DISKNAME}":"/dev/nvme1n1",
			"{#DISKCMD}":"/dev/nvme1n1",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}
```

without nvme disks:
```
{
	"data":[

		{
			"{#DISKMODEL}":"",
			"{#DISKSN}":"",
			"{#DISKNAME}":"/dev/hdd",
			"{#DISKCMD}":"/dev/hdd -d ata",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"2"
		}
	]
}
```

Reverting the following:
```
-    for (`$smartctl_cmd --scan-open`) {
+    for (`$smartctl_cmd --scan-open`.
+         `$smartctl_cmd --scan-open -dnvme`) {
```
gives correct output for both nvme and ordinary disks.

v-zhuravlev, 2019-05-17:
@nerijus , sorry what do you by 'with nvme disks'? On host with nvme disks?

nerijus, 2019-05-17:
Yes, I mean host with nvme disks.
```
# smartctl --scan-open -d nvme
/dev/nvme0 -d nvme # /dev/nvme0, NVMe device
/dev/nvme1 -d nvme # /dev/nvme1, NVMe device
```
```
# smartctl --scan-open
# scan_smart_devices: glob(3) aborted matching pattern /dev/discs/disc*
```

nerijus, 2019-05-17:
BTW, with the above smartctl-disks-discovery.pl on one server I have:
```
# ./smartctl-disks-discovery.pl
{
	"data":[

		{
			"{#DISKMODEL}":"WDC WD1003FBYZ-010FB0",
			"{#DISKSN}":"WD-WCAW37577909",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"WDC WD1003FBYZ-010FB0",
			"{#DISKSN}":"WD-WCAW37372345",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		}
	]
}
```
but zabbix displays the 2nd disk 2 times:
[image]

Skunnyk, 2019-06-11:
Hi, I have some nvme disks where I need to pass `/dev/nvme0` instead of `/dev/nvme0n1` to `uSSD.get` (ssmartctl 6.6/debian stretch, and see https://www.smartmontools.org/ticket/1134)

v-zhuravlev, 2019-06-12:
@Skunnyk , you may try to workaround this by using static discovery:
https://github.com/v-zhuravlev/zbx-smartctl#about-static-discovery

v-zhuravlev, 2019-06-12:
@brosky, @nerijus - updated script here: https://github.com/v-zhuravlev/zbx-smartctl/pull/92/commits/9905f737baaa28f7406a4878957720b632ef8aa4
It would be great if you can test with and without nvme-cli.

nerijus, 2019-06-13:
It works the same with and without nvme-cli.
Old script output:
```
{
        "data":[

                {
                        "{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
                        "{#DISKSN}":"S3W8NX0M418376",
                        "{#DISKNAME}":"/dev/nvme0n1",
                        "{#DISKCMD}":"/dev/nvme0n1",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                },
                {
                        "{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
                        "{#DISKSN}":"S3W8NX0M418304",
                        "{#DISKNAME}":"/dev/nvme1n1",
                        "{#DISKCMD}":"/dev/nvme1n1",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                }
        ]
}
```
New script output:
```
{
        "data":[

                {
                        "{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
                        "{#DISKSN}":"S3W8NX0M418376",
                        "{#DISKNAME}":"/dev/nvme0",
                        "{#DISKCMD}":"/dev/nvme0 -d nvme",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                },
                {
                        "{#DISKMODEL}":"SAMSUNG MZVLB512HAJQ-00000",
                        "{#DISKSN}":"S3W8NX0M418304",
                        "{#DISKNAME}":"/dev/nvme1",
                        "{#DISKCMD}":"/dev/nvme1 -d nvme",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                }
        ]
}
```

v-zhuravlev, 2019-06-13:
@nerijus , thats great, so `nvme-cli` is not a requirement anymore. Do you have any issues left with NVMe disks with new script?

nerijus, 2019-06-13:
It seems ok so far, thank you. BTW, ID 5 Reallocated sectors count is not supported on my nvme disks, do nvme disks support this attribute at all?

v-zhuravlev, 2019-06-13:
Yes, there is no such attribute for NVMe at all AFAIK

nerijus, 2019-06-13:
Could it be disabled automatically for nvme disks?

v-zhuravlev, 2019-06-13:
no, not really

Skunnyk, 2019-06-13:
The last version of `nvme_discovery` branch works perfectly for me on differents nvme disks, ssd and hdd!

v-zhuravlev, 2019-06-13:
@Skunnyk, glad that it works for you

v-zhuravlev, 2019-06-13:
closing it now

PR #86 (2019-04-24): Nvme
fixes #85
closed 2019-06-13 by v-zhuravlev