v-zhuravlev/zbx-smartctl #94: Discovery rules Value should be a JSON object

zetheroo, 2019-05-09:
I saw there was a closed Issue for this, and I commented there, but am not sure if anyone is notified about comments on closed issues.

I am experiencing this issue on a Debian Stretch (PVE) system running zabbix_agent 4.0.7

root@proxsat:/etc/zabbix/scripts# ./smartctl-disks-discovery.pl 
{
	"data":[

		{
			"{#DISKMODEL}":"WDC WD1003FZEX-00K3CA0",
			"{#DISKSN}":"WD-WCC6Y0NYF3E7",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"WDC WD1003FZEX-00K3CA0",
			"{#DISKSN}":"WD-WCC6Y4KD46L2",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"Samsung SSD 860 PRO 1TB",
			"{#DISKSN}":"S42NNF0K501976K",
			"{#DISKNAME}":"/dev/sdc",
			"{#DISKCMD}":"/dev/sdc -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"Samsung SSD 860 PRO 1TB",
			"{#DISKSN}":"S42NNF0K501910F",
			"{#DISKNAME}":"/dev/sdd",
			"{#DISKCMD}":"/dev/sdd -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}

Also on a FreeNAS system with zabbix_agent 3.2. So far it's only working on the Zabbix Server (Ubuntu 18.04) itself.

Update: On another system (Ubuntu 18.04) with zabbix_agent 4.0.7 it works without a hitch.

[root@hive /etc/zabbix/scripts]# ./smartctl-disks-discovery.pl
{
	"data":[

		{
			"{#DISKMODEL}":"WDC WD30EZRX-00AZ6B0",
			"{#DISKSN}":"WD-WCC070364464",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"WDC WD30EZRX-00DC0B0",
			"{#DISKSN}":"WD-WCC1T0896736",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"0"
		}
	]
}
So far the only systems it works on are Ubuntu systems.

v-zhuravlev, 2019-05-13:
don't test as `root`, test as `zabbix` user. Or enable debug Logging on agent - you will probably see there why collected value is not a json.

zetheroo, 2019-05-23:
Ok, so we sorted out the permissions issue and got this working 100% with Debian (ProxmoxVE), but for FreeNAS it took a whole lot more digging to figure out the solution ... which was  ... 

adding

```
zabbix ALL=(ALL) NOPASSWD: /usr/local/sbin/smartctl
zabbix ALL=(ALL) NOPASSWD: /usr/local/etc/zabbix34/scripts/smartctl-disks-discovery.pl
```

to visudo.

v-zhuravlev, 2019-05-23:
Isn't visudo mentioned in README?:
```
Cmnd_Alias SMARTCTL = /usr/sbin/smartctl
Cmnd_Alias SMARTCTL_DISCOVERY = /etc/zabbix/scripts/smartctl-disks-discovery.pl
zabbix ALL= (ALL) NOPASSWD: SMARTCTL, SMARTCTL_DISCOVERY
Defaults!SMARTCTL !logfile, !syslog, !pam_session
Defaults!SMARTCTL_DISCOVERY !logfile, !syslog, !pam_session
```

v-zhuravlev, 2019-05-23:
Glad that you sorted it that out

tschaerni, 2019-05-29:
Hi,
@zetheroo 
I have also the same problem like you under proxmox, would you like to tell me whats your solution?

Machine:
Debian Stretch (proxmox, latest updates)
Zabbix Server, Proxy and Agent versions: 4.0.8

Output from zabbix_get from the proxy:
```
root@dvalin-zabbix-proxy:/# zabbix_get -s192.168.21.10 -p10050 -kuHDD.discovery
{
        "data":[

                {
                        "{#DISKMODEL}":"Corsair Force GT",
                        "{#DISKSN}":"1151820200000689052E",
                        "{#DISKNAME}":"/dev/sda",
                        "{#DISKCMD}":"/dev/sda -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                },
                {
                        "{#DISKMODEL}":"SAMSUNG SSD PM830 2.5" 7mm 256GB",
                        "{#DISKSN}":"S0TZNSAD115314",
                        "{#DISKNAME}":"/dev/sdb",
                        "{#DISKCMD}":"/dev/sdb -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                }
        ]
}
```
If my understanding is correct, the discovery script is running fine, so no permission issues, but I also get the error "Value should be a JSON object" in the discovery rules...

Hoping for a solution, because I don't really know how to debug it further :/

v-zhuravlev, 2019-06-12:
Thank you, I added escaping of quotes in #100

closed 2019-06-12 by v-zhuravlev