v-zhuravlev/zbx-smartctl #96: Some issue with ssd regexp

lexa500, 2019-05-19:
1164:20190519:203456.466 error reason for "some_host_name:uSSD["/dev/sda -d sat",UDMA_CRC_Error]" changed: Item preprocessing step #1 failed: cannot perform regular expression match on value "smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-12-pve] (local build)

YogSottot, 2019-08-16:
The same:

```
Item preprocessing step #1 failed: cannot perform regular expression match: pattern does not match, type "string", value "smartctl 6.5 2016-05-07 r4318 [x86_64-linux-3.10.0-957.27.2.el7.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME FLAG VALUE WORST THRESH TYPE UPDATED WHEN_FAILED RAW_VALUE
5 Reallocated_Sector_Ct 0x0033 100 100 010 Pre-fail Always - 0
9 Power_On_Hours 0x0032 096 096 000 Old_age Always - 17592
12 Power_Cycle_Count 0x0032 099 099 000 Old_age Always - 26
170 Unused_Rsvd_Blk_Ct_Chip 0x0032 100 100 010 Old_age Always - 0
171 Program_Fail_Count_Chip 0x0032 100 100 010 Old_age Always - 0
172 Erase_Fail_Count_Chip 0x0032 100 100 010 Old_age Always - 0
173 Wear_Leveling_Count 0x0033 076 076 005 Pre-fail Always - 295
174 Unexpect_Power_Loss_Ct 0x0032 099 099 000 Old_age Always - 20
178 Used_Rsvd_Blk_Cnt_Chip 0x0013 100 100 010 Pre-fail Always - 0
180 Unused_Rsvd_Blk_Cnt_Tot 0x0013 100 100 010 Pre-fail Always - 1648
184 End-to-End_Error 0x0033 100 100 097 Pre-fail Always - 0
187 Uncorrectable_Error_Cnt 0x0032 100 100 000 Old_age Always - 0
194 Temperature_Celsius 0x0032 074 056 000 Old_age Always - 26
199 CRC_Error_Count 0x003e 100 100 000 Old_age Always - 0
233 Media_Wearout_Indicator 0x0013 075 075 000 Pre-fail Always - 12652816
241 Total_LBAs_Written 0x0032 099 099 000 Old_age Always - 130656
242 Total_LBAs_Read 0x0032 099 099 000 Old_age Always 
```

v-zhuravlev, 2019-11-01:
Sorry, that's not enough to troubleshoot. :( Perhaps, which version of template do you use and what is the output of smarctl cli?