v-zhuravlev/zbx-smartctl #97: Discovery Script does not escape double quotation marks

tschaerni, 2019-05-30:
Hi,
@zetheroo 
I have also the same problem like you under proxmox, would you like to tell me whats your solution?

Machine:
Debian Stretch (proxmox, latest updates)
Zabbix Server, Proxy and Agent versions: 4.0.8

Output from zabbix_get from the proxy:
```
root@dvalin-zabbix-proxy:/# zabbix_get -s192.168.21.10 -p10050 -kuHDD.discovery
{
        "data":[

                {
                        "{#DISKMODEL}":"Corsair Force GT",
                        "{#DISKSN}":"1151820200000689052E",
                        "{#DISKNAME}":"/dev/sda",
                        "{#DISKCMD}":"/dev/sda -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                },
                {
                        "{#DISKMODEL}":"SAMSUNG SSD PM830 2.5" 7mm 256GB",
                        "{#DISKSN}":"S0TZNSAD115314",
                        "{#DISKNAME}":"/dev/sdb",
                        "{#DISKCMD}":"/dev/sdb -d sat",
                        "{#SMART_ENABLED}":"1",
                        "{#DISKTYPE}":"1"
                }
        ]
}
```
If my understanding is correct, the discovery script is running fine, so no permission issues, but I also get the error "Value should be a JSON object" in the discovery rules...

Hoping for a solution, because I don't really know how to debug it further :/

**EDIT :** zabbix_get -s 192.168.21.10 -k uHDD.get[/dev/sda] works, and returns the S.M.A.R.T Attributes
**EDIT nr.2 :** I think I found a bug in the discovery script which causes my problems. I did run json output from above through https://jsonlint.com/ and it found a problem in line 13:
```                        "{#DISKMODEL}":"SAMSUNG SSD PM830 2.5" 7mm 256GB",```
It looks like the quotation mark from 2.5" is interpreted as the end of the datafield and because there is no following comma, its not valid json...
I don't know perl, so I can't really try to fix it, but I hope that @v-zhuravlev maybe can fix that :/

_Originally posted by @tschaerni in https://github.com/v-zhuravlev/zbx-smartctl/issues/94#issuecomment-497149349_

tschaerni, 2019-05-30:
Small update:
I just wrote a wrapper script in bash to escape the one double quotation mark:
```
#!/bin/bash
/etc/zabbix/scripts/smartctl-disks-discovery.pl.bak | sed 's/2.5\"/2.5\\\"/'
exit 0
```
After that, the discovery rule works and it populated the items and triggers, the Items also report correctly. Atleast most of the values are collected correctly.
It looks like I have some special SSDs here, because some of the IDs does not match, or aren't collected by the firmware of the SSD... ("5 Retired_Block_Count" instead of "5 Reallocated_Sector_Ct").

If interested I post the output of smartctl here:
```
=== START OF INFORMATION SECTION ===
Model Family:     SandForce Driven SSDs
Device Model:     Corsair Force GT
Serial Number:    1151820200000689052E
LU WWN Device Id: 0 000000 000000000
Firmware Version: 1.3.3
User Capacity:    120,034,123,776 bytes [120 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ATA8-ACS, ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu May 30 03:23:22 2019 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

SMART Attributes Data Structure revision number: 10
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   094   094   050    Pre-fail  Always       -       0/178154236
  5 Retired_Block_Count     0x0033   100   100   003    Pre-fail  Always       -       0
  9 Power_On_Hours_and_Msec 0x0032   047   047   000    Old_age   Always       -       47174h+11m+42.620s
 12 Power_Cycle_Count       0x0032   099   099   000    Old_age   Always       -       1559
171 Program_Fail_Count      0x0032   000   000   000    Old_age   Always       -       0
172 Erase_Fail_Count        0x0032   000   000   000    Old_age   Always       -       0
174 Unexpect_Power_Loss_Ct  0x0030   000   000   000    Old_age   Offline      -       195
177 Wear_Range_Delta        0x0000   000   000   000    Old_age   Offline      -       3
181 Program_Fail_Count      0x0032   000   000   000    Old_age   Always       -       0
182 Erase_Fail_Count        0x0032   000   000   000    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
194 Temperature_Celsius     0x0022   035   060   000    Old_age   Always       -       35 (Min/Max 12/60)
195 ECC_Uncorr_Error_Count  0x001c   120   120   000    Old_age   Offline      -       0/178154236
196 Reallocated_Event_Count 0x0033   100   100   003    Pre-fail  Always       -       0
201 Unc_Soft_Read_Err_Rate  0x001c   120   120   000    Old_age   Offline      -       0/178154236
204 Soft_ECC_Correct_Rate   0x001c   120   120   000    Old_age   Offline      -       0/178154236
230 Life_Curve_Status       0x0013   100   100   000    Pre-fail  Always       -       100
231 SSD_Life_Left           0x0013   100   100   010    Pre-fail  Always       -       0
233 SandForce_Internal      0x0000   000   000   000    Old_age   Offline      -       15216
234 SandForce_Internal      0x0032   000   000   000    Old_age   Always       -       10924
241 Lifetime_Writes_GiB     0x0032   000   000   000    Old_age   Always       -       10924
242 Lifetime_Reads_GiB      0x0032   000   000   000    Old_age   Always       -       21928
```
and the second one:

```
=== START OF INFORMATION SECTION ===
Device Model:     SAMSUNG SSD PM830 2.5" 7mm 256GB
Serial Number:    S0TZNSAD115314
LU WWN Device Id: 5 002538 043584d30
Firmware Version: CXM03D1Q
User Capacity:    256,060,514,304 bytes [256 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   ATA8-ACS T13/1699-D revision 4c
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu May 30 03:22:53 2019 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0033   100   100   010    Pre-fail  Always       -       0
  9 Power_On_Hours          0x0032   097   097   000    Old_age   Always       -       10412
 12 Power_Cycle_Count       0x0032   098   098   000    Old_age   Always       -       1848
175 Program_Fail_Count_Chip 0x0032   100   100   010    Old_age   Always       -       0
176 Erase_Fail_Count_Chip   0x0032   100   100   010    Old_age   Always       -       0
177 Wear_Leveling_Count     0x0013   095   095   010    Pre-fail  Always       -       180
178 Used_Rsvd_Blk_Cnt_Chip  0x0013   094   094   010    Pre-fail  Always       -       216
179 Used_Rsvd_Blk_Cnt_Tot   0x0013   095   095   010    Pre-fail  Always       -       388
180 Unused_Rsvd_Blk_Cnt_Tot 0x0013   095   095   010    Pre-fail  Always       -       7676
181 Program_Fail_Cnt_Total  0x0032   100   100   010    Old_age   Always       -       0
182 Erase_Fail_Count_Total  0x0032   100   100   010    Old_age   Always       -       0
183 Runtime_Bad_Block       0x0013   100   100   010    Pre-fail  Always       -       0
[... cut ...]
195 Hardware_ECC_Recovered  0x001a   200   200   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0030   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x003e   253   253   000    Old_age   Always       -       12
232 Available_Reservd_Space 0x0013   094   094   000    Pre-fail  Always       -       3816
241 Total_LBAs_Written      0x0032   099   099   000    Old_age   Always       -       10506175841
242 Total_LBAs_Read         0x0032   099   099   000    Old_age   Always       -       19339140549
```

v-zhuravlev, 2019-06-12:
Thanks for reporting I added simple sub to escape quotes in json string

closed 2019-06-12 by v-zhuravlev

tschaerni, 2019-06-13:
Nice! thanks for the update 👍

PR #100 (2019-06-12): escape " and \ in json
fixes #97