v-zhuravlev/zbx-smartctl #98: Permanently lit SMART trigger status is not OK

AntonAndreevichMoroz, 2019-05-31:
Trigger SMART status is not OK constantly in the Problem state for all disks.

AntonAndreevichMoroz, 2019-05-31:
But in fact, all results get PASSED. Why does the trigger fire?

v-zhuravlev, 2019-05-31:
please show me trigger expression and what is the value that you have?

v-zhuravlev, 2019-05-31:
Is client Windows or Linux?

v-zhuravlev, 2019-05-31:
My guess is that there is a whitespace character in the output. 
Try changing regex in preprocessing step for something like this?
`(?:SMART overall-health self-assessment test result:|SMART Health Status:) +\b([\S ]+)`
or if this doesn't help. maybe even this:
`(?:SMART overall-health self-assessment test result:|SMART Health Status:) +\b([\w ]+)`

AntonAndreevichMoroz, 2019-06-03:
Windows client.
I eventually made these conditions for the trigger. Unfortunately, I don’t know how to imitate 100% of a poor SMART state, but changing the text in a macro gives the correct triggering effect. Please give your expert opinion how much it is correct. Thank you in advance.
{Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uSSD.health["{#DISKCMD}"].iregexp({$SMARTCTL_OK_STATUS:"SATA"},#1)}=0 and {Template_HDD_SMARTMONTOOLS_2_WITH_LLD:uSSD.health["{#DISKCMD}"].iregexp({$SMARTCTL_OK_STATUS:"SAS"},#1)}=0

v-zhuravlev, 2019-06-13:
Can you try to revert trigger expression back and the line I suggested in this item preprocessing?
`(?:SMART overall-health self-assessment test result:|SMART Health Status:) +\b([\S ]+)`

gijoe88, 2019-07-03:
The new regexp (with \S) solved the issue for me. I don't think it was linked to there being a space in the result, but to "line endings" of windows having "\r\n", the \r being caught by "." when on unix server.

closed 2019-07-11 by v-zhuravlev