v-zhuravlev/zbx-smartctl #103: Cannot import template in zabbix 4.2

Jensilein, 2019-06-16:
Hello,

I tried to import the 3.4 template in zabbix 4.2 but I get the failure message

> Cannot read XML: (41) Specification mandates value for attribute data-pjax-transient [Line: 44 | Column: 92].

How can I make it work with zabbix 4.2? Would be great to find a solution.

Best regards

v-zhuravlev, 2019-06-17:
This template works in 4.2 without any issues. Please make sure you are using the latest template from this repo.
[image]

Please reopen if still having problems.

closed 2019-06-17 by v-zhuravlev

Jensilein, 2019-06-17:
Hi Vitaly,

Thanks for your swift reply. Unfortunately it doesn‘t work for me. I always get the above mentioned error message. Maybe it is because I have zabbix running in a docker on ky Synology NAS? Or that doesn‘t matter at all? But how can I solve the problem?

Thanks and best regards.