v-zhuravlev/zbx-smartctl #104: This template works in 4.2 without any issues. Please make sure you are using the latest template from this repo.

Jensilein, 2019-06-18:
This template works in 4.2 without any issues. Please make sure you are using the latest template from this repo.
[image]

Please reopen if still having problems.

_Originally posted by @v-zhuravlev in https://github.com/v-zhuravlev/zbx-smartctl/issues/103#issuecomment-502598408_

Jensilein, 2019-06-18:
Hi Vitaly,

Thanks for your swift reply. Unfortunately it doesn‘t work for me. I always get the above mentioned error message. Maybe it is because I have zabbix running in a docker on my Synology NAS or doesn’t that matter at all? But how can I solve the problem, do you have any idea?

Thanks and best regards.

Jensilein, 2019-06-18:
Hello Vitaly, problem solved. Have downloaded the files again and now the import worked.

v-zhuravlev, 2019-06-18:
Glad to hear that:)

Отправлено с телефона

closed 2019-06-19 by v-zhuravlev