v-zhuravlev/zbx-smartctl #108: Error on linux: Value should be a JSON object.

TiratoreA, 2019-07-04:
[image]
smartctl --scan-open
/dev/sda -d scsi # /dev/sda, SCSI device
/dev/sdb -d sat # /dev/sdb [SAT], ATA device
/dev/bus/0 -d megaraid,0 # /dev/bus/0 [megaraid_disk_00], SCSI device
/dev/bus/0 -d megaraid,1 # /dev/bus/0 [megaraid_disk_01], SCSI device
/dev/bus/0 -d megaraid,2 # /dev/bus/0 [megaraid_disk_02], SCSI device
/dev/bus/0 -d megaraid,3 # /dev/bus/0 [megaraid_disk_03], SCSI device
/dev/bus/0 -d megaraid,4 # /dev/bus/0 [megaraid_disk_04], SCSI device
/dev/bus/0 -d megaraid,5 # /dev/bus/0 [megaraid_disk_05], SCSI device
/dev/bus/0 -d megaraid,6 # /dev/bus/0 [megaraid_disk_06], SCSI device
/dev/bus/0 -d megaraid,7 # /dev/bus/0 [megaraid_disk_07], SCSI device

closed 2019-07-05 by TiratoreA

Hyrla, 2020-03-23:
Hey, I have the same problem, how did you solve it ? Thank you

TiratoreA, 2020-03-23:
Copy zabbix_smartctl.conf to /etc/zabbix/zabbix_agentd.d
Copy script smartctl-disks-discovery.pl to /etc/zabbix/scripts
Then run
 chown zabbix:zabbix /etc/zabbix/scripts/smartctl-disks-discovery.pl
chmod u+x /etc/zabbix/scripts/smartctl-disks-discovery.pl
chown root.sys /usr/sbin/smartctl
usermod -a -G sys zabbix
chmod u+s /usr/sbin/smartctl
chown root.sys /usr/bin/sg_scan
chmod u+s /usr/bin/sg_scan