v-zhuravlev/zbx-smartctl #110: Не может проверить данные за raid-контроллером Adaptec

hadovulent, 2019-08-03:
PS C:\zabbix\bin\win64> .\zabbix_agentd.exe -c C:\zabbix\conf\zabbix_agentd.win.conf -t uHDD.get["/dev/sda -d aacraid,1,0,0"]
uHDD.get[/dev/sda -d aacraid,1,0,0]           [t|smartctl 7.0 2018-12-30 r4883 [x86_64-w64-mingw32-2019] (sf-7.0-1)
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org
/dev/sda: Unknown device type 'aacraid'
=======> VALID ARGUMENTS ARE: ata, scsi[+TYPE], nvme[,NSID], sat[,auto][,N][+TYPE], usbcypress[,X], usbjmicron[,p][,x][,N], usbprolific, usbsunplus, sntjmicron[,NSID], intelliprop,N[+TYPE], aacraid,H,L,ID, areca,N[/E], auto, test <=======
Use smartctl -h to get a usage summary]

c:\Program Files\smartmontools\bin>smartctl /dev/sda -i -d aacraid,1,0,0
smartctl 7.0 2018-12-30 r4883 [x86_64-w64-mingw32-2019] (sf-7.0-1)
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate IronWolf
Device Model:     ST2000VN004-2E4164
Serial Number:    Z526W3NW
LU WWN Device Id: 5 000c50 0b363bc45
Firmware Version: SC60
User Capacity:    2 000 398 934 016 bytes [2,00 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    5900 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-2, ACS-3 T13/2161-D revision 3b
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Sat Aug 03 08:15:29 2019 RTZ
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

kosfango, 2019-08-03:
I  use this:  https://github.com/YSmetana/raid_arcconf_zabbix_lld
for Adaptec raid monitoring

hadovulent, 2019-08-03:
This is for linux system. And i need for windows.
https://github.com/mykolq/zabbix_adaptec_template - This one I use for SAS drives.

kosfango, 2019-08-03:
I think you can try to run python in Windows with minimal the script editing. However, arcconf output can be different in Windows.

dudkitt, 2019-12-11:
I see - In uHDD[ ] not allowed symbol ","

hadovulent, 2019-12-13:
And where exactly is there no symbol?
If you succeed, then write your command

hadovulent, 2019-12-13:
> I see - in uHDD.get[ ] not allowed symbol ","

And where exactly is there no symbol?
If you succeed, then write your command