v-zhuravlev/zbx-smartctl #111: Error during script execution on windows host

Murtadin, 2019-08-06:
Hello dear. On one host it works fine without errors (Win2k8r2 x64 enterprise). But on three more servers i have got such error:

Win2k8R2 x64 standart:
```
PS C:\Program Files\Zabbix Agent> .\smartctl-disks-discovery.ps1
{
 "data":[
You cannot call a method on a null-valued expression.
At C:\Program Files\Zabbix Agent\smartctl-disks-discovery.ps1:34 char:73
+     $disk_name = $smart_scanresult.Substring(0,$smart_scanresult.IndexOf <<<< (" "))
    + CategoryInfo          : InvalidOperation: (IndexOf:String) [], RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

         {
                "{#DISKSN}":"",
                "{#DISKMODEL}":"Intel Raid 1 Volume",
                "{#DISKNAME}":"/dev/sda",
                "{#DISKCMD}":"/dev/sda -dscsi",
                "{#SMART_ENABLED}":"0",
                "{#DISKTYPE}":"2"
         },
         {
                "{#DISKSN}":"WD-WCC2E4ST07T5",
                "{#DISKMODEL}":"WDC WD5000AAKX-22ERMA0",
                "{#DISKNAME}":"/dev/csmi0,0",
                "{#DISKCMD}":"/dev/csmi0,0 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
         },
         {
                "{#DISKSN}":"WD-WCC2E1KY3XPD",
                "{#DISKMODEL}":"WDC WD5000AAKX-22ERMA0",
                "{#DISKNAME}":"/dev/csmi0,1",
                "{#DISKCMD}":"/dev/csmi0,1 -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
         },
         {
                "{#DISKSN}":"",
                "{#DISKMODEL}":"",
                "{#DISKNAME}":"",
                "{#DISKCMD}":" ",
                "{#SMART_ENABLED}":"0",
                "{#DISKTYPE}":"2"
         }
 ]
}
PS C:\Program Files\Zabbix Agent>
```

WIndows Server 2008 x64 R2 Datacenter:

```
PS C:\Program Files\Zabbix Agent> .\smartctl-disks-discovery.ps1
{
 "data":[
Нельзя вызвать метод для выражения со значением NULL.
C:\Program Files\Zabbix Agent\smartctl-disks-discovery.ps1:34 знак:73
+     $disk_name = $smart_scanresult.Substring(0,$smart_scanresult.IndexOf <<<< (" "))
    + CategoryInfo          : InvalidOperation: (IndexOf:String) [], RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

         {
                "{#DISKSN}":"50026B724402C542",
                "{#DISKMODEL}":"KINGSTON SH103S3120G",
                "{#DISKNAME}":"/dev/sda",
                "{#DISKCMD}":"/dev/sda -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"1"
         },
         {
                "{#DISKSN}":"Z1E71R8W",
                "{#DISKMODEL}":"ST2000DM001-1CH164",
                "{#DISKNAME}":"/dev/sdb",
                "{#DISKCMD}":"/dev/sdb -data",
                "{#SMART_ENABLED}":"1",
                "{#DISKTYPE}":"0"
         },
         {
                "{#DISKSN}":"",
                "{#DISKMODEL}":"",
                "{#DISKNAME}":"",
                "{#DISKCMD}":" ",
                "{#SMART_ENABLED}":"0",
                "{#DISKTYPE}":"2"
         }
 ]
}
PS C:\Program Files\Zabbix Agent>
```

Can you pls help me how to get it working? Thank you in advance.

avalishev, 2019-08-30:
This is powershell error, install WMF 5.1 [https://docs.microsoft.com/powershell/wmf/setup/install-configure](url)

Murtadin, 2019-08-31:
Thank you. Now all works as expected.

closed 2019-09-01 by Murtadin