v-zhuravlev/zbx-smartctl #112: Error importing Template with Zabbix 3.4.15

schneidr, 2019-09-09:
Trying to import the 3.4 template results in this error message:

> Import failed
> Details
> - System error occurred. Please contact Zabbix administrator.
> - SQL statement execution has failed "INSERT INTO items (name,type,snmp_community,snmp_oid,delay,history,trends,status,value_type,units,snmpv3_contextname,snmpv3_securityname,snmpv3_securitylevel,snmpv3_authprotocol,snmpv3_authpassphrase,snmpv3_privprotocol,snmpv3_privpassphrase,params,ipmi_sensor,authtype,username,password,publickey,privatekey,port,description,inventory_link,logtimefmt,jmx_endpoint,key_,trapper_hosts,hostid,master_itemid,flags,itemid) VALUES ('{#DISKNAME}: ID 190/194 Temperature','18','','','0','7d','90d','0','3','°С','','','0','0','','0','','','','0','','','','','','Current drive temperature.\nSATA: Parses value from attribute ID 194 or ID 190.\nSAS: Supported\nNVMe: Supported','0','','','uHDD[\"{#DISKCMD}\",Temperature_Celsius]','','10277','29562','2','29572')".

No error message in `/var/log/zabbix/zabbix_server.log`.

Running that SQL query manually in phpMyAdmin results in this error message:

> #1366 - Incorrect string value: '\xD0\xA1' for column 'units' at row 1

v-zhuravlev, 2019-11-01:
What encoding do you have in the DB? Is it UTF8? This line is the problem:
`<units>°С</units>`

ntherning, 2020-01-02:
I had the same issue. My `items` table was using `latin1` as default character set. After running
```sql
alter table items convert to character set utf8
```
the template could be imported successfully.

Wohlraj, 2020-02-27:
Hi, encountered this issue as well.
This is because the template has the letter С in Cyrillic (&#1057;) instead of C in Latin (&#67;).
Would you accept a pull request to switch to Latin?

aphex3k, 2020-04-04:
I created a PR for the version 3.4 template: https://github.com/v-zhuravlev/zbx-smartctl/pull/134