v-zhuravlev/zbx-smartctl #115: Discovery rule stop working after adding second hdd

dudkitt, 2019-09-20:
Windows7 zabbix-agent 4.0.11, zabbix-server 4.0.11 . When I add second disk to host- discovery rule stop working.
Before

> PS C:\Windows\system32> C:\Program Files\Zabbix\smartctl-disks-discovery.ps1
> {
> "data":[
> {
> "{#DISKSN}":"5VM0T8KV",
> "{#DISKMODEL}":"ST3500410AS",
> "{#DISKNAME}":"/dev/sda",
> "{#DISKCMD}":"/dev/sda -data",
> "{#SMART_ENABLED}":"1",
> "{#DISKTYPE}":"0"
> }
> ]
> }

after

> PS C:\Windows\system32> C:\Program Files\Zabbix\smartctl-disks-discovery.ps1
> 5VM0T8KV
> {
> 
> 
> "data":[
> Нельзя вызвать метод для выражения со значением NULL.
> C:\Program Files\Zabbix\smartctl-disks-discovery.ps1:34 знак:73
> + $disk_name = $smart_scanresult.Substring(0,$smart_scanresult.IndexOf <<<< (" "))
> + CategoryInfo : InvalidOperation: (IndexOf:String) [], RuntimeException
> + FullyQualifiedErrorId : InvokeMethodOnNull
> ,
> {
> "{#DISKSN}":"WD-WCC4M2742388",
> "{#DISKMODEL}":"WDC WD20EZRX-00D8PB0",
> "{#DISKNAME}":"/dev/sdb",
> "{#DISKCMD}":"/dev/sdb -data",
> "{#SMART_ENABLED}":"1",
> "{#DISKTYPE}":"0"
> },
> 
> 
> {
> "{#DISKSN}":"",
> "{#DISKMODEL}":"",
> "{#DISKNAME}":"",
> "{#DISKCMD}":" ",
> "{#SMART_ENABLED}":"0",
> "{#DISKTYPE}":"2"
> }
> ]
> }_______________________________________________________________________________________________________________________________________
> PS C:\Windows\system32> C:\Program Files\Zabbix\smartctl-disks-discovery.ps1
> 5VM0T8KVWD-WCC4M2742388
> {
> "data":[
> Нельзя вызвать метод для выражения со значением NULL.
> C:\Program Files\Zabbix\smartctl-disks-discovery.ps1:34 знак:73
> + $disk_name = $smart_scanresult.Substring(0,$smart_scanresult.IndexOf <<<< (" "))
> + CategoryInfo : InvalidOperation: (IndexOf:String) [], RuntimeException
> + FullyQualifiedErrorId : InvokeMethodOnNull

v-zhuravlev, 2019-11-01:
What is the second disk? What is the output of `smarctl --scan-open` and `smartctl -x <second_disk>`?

dudkitt, 2019-11-01:
C:\Users\user10>smartctl --scan-open
/dev/sda -d ata # /dev/sda, ATA device
/dev/sdb -d ata # /dev/sdb, ATA device

C:\Users\user10>smartctl -x /dev/sdb
smartctl 7.0 2018-12-30 r4883 [x86_64-w64-mingw32-win7-sp1] (sf-7.0-1)
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org

Warning: Limited functionality due to missing admin rights
=== START OF INFORMATION SECTION ===
Model Family:     Western Digital Green
Device Model:     WDC WD20EZRX-00D8PB0
Serial Number:    2020202057202d444357
Firmware Version: 80.00A80
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   [No Information Found]
Local Time is:    Fri Nov 01 09:31:43 2019 RTZ
SMART support is: Available - device has SMART capability.
                  Enabled status cached by OS, trying SMART RETURN STATUS cmd.
SMART support is: Enabled
AAM feature is:   Unavailable
APM feature is:   Unavailable
Rd look-ahead is: Unavailable
Write cache is:   Unavailable
DSN feature is:   Unavailable
ATA Security is:  Unavailable
Wt Cache Reorder: Unavailable

Read SMART Thresholds failed: Function not implemented

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x82) Offline data collection activity
                                        was completed without error.
                                        Auto Offline Data Collection: Enabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (25200) seconds.
Offline data collection
capabilities:                    (0x7b) SMART execute Offline immediate.
                                        Auto Offline data collection on/off support.
                                        Suspend Offline collection upon new
                                        command.
                                        Offline surface scan supported.
                                        Self-test supported.
                                        Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        No General Purpose Logging support.
Short self-test routine
recommended polling time:        (   2) minutes.
Extended self-test routine
recommended polling time:        ( 255) minutes.
Conveyance self-test routine
recommended polling time:        (   5) minutes.

SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAGS    VALUE WORST THRESH FAIL RAW_VALUE
  1 Raw_Read_Error_Rate     POSR-K   163   160   ---    -    65537
  3 Spin_Up_Time            POS--K   174   170   ---    -    4283
  4 Start_Stop_Count        -O--CK   097   097   ---    -    3101
  5 Reallocated_Sector_Ct   PO--CK   200   200   ---    -    0
  7 Seek_Error_Rate         -OSR-K   200   200   ---    -    0
  9 Power_On_Hours          -O--CK   047   047   ---    -    39360
 10 Spin_Retry_Count        -O--CK   100   100   ---    -    0
 11 Calibration_Retry_Count -O--CK   100   100   ---    -    0
 12 Power_Cycle_Count       -O--CK   099   099   ---    -    1413
192 Power-Off_Retract_Count -O--CK   200   200   ---    -    34
193 Load_Cycle_Count        -O--CK   197   197   ---    -    9135
194 Temperature_Celsius     -O---K   116   107   ---    -    31
196 Reallocated_Event_Count -O--CK   200   200   ---    -    0
197 Current_Pending_Sector  -O--CK   200   200   ---    -    39
198 Offline_Uncorrectable   ----CK   200   200   ---    -    4
199 UDMA_CRC_Error_Count    -O--CK   200   200   ---    -    0
200 Multi_Zone_Error_Rate   ---R--   192   192   ---    -    3236
                            ||||||_ K auto-keep
                            |||||__ C event count
                            ||||___ R error rate
                            |||____ S speed/performance
                            ||_____ O updated online
                            |______ P prefailure warning

Read SMART Log Directory failed: Function requires admin rights
General Purpose Log Directory not supported
SMART Extended Comprehensive Error Log (GP Log 0x03) not supported
Read SMART Error Log failed: Function requires admin rights
SMART Extended Self-test Log (GP Log 0x07) not supported
Read SMART Self-test Log failed: Function requires admin rights
Read SMART Selective Self-test Log failed: Function requires admin rights
SCT Commands not supported
Device Statistics (GP/SMART Log 0x04) not supported
Pending Defects log (GP Log 0x0c) not supported
SATA Phy Event Counters (GP Log 0x11) not supported