v-zhuravlev/zbx-smartctl #120: Disktype incorrect definition

vopatrushev, 2019-10-31:
Добрый день!
Некоторые диски определились как "{#DISKTYPE}":"2", и соответственно сервер получает с них информацию, но не передает в мониторинг. 
Путем поисков в smartctl -x определено, что это происходит на дисках с **ATA Standard is:  Not recognized.**
Можно ли вручную указывать тип диска?

v-zhuravlev, 2019-11-01:
Нет сейчас такой возможности нет.
Можете показать полный вывод extended? Там нет других более явных признаком что это SSD/HDD? 
Если есть, то можно добавить регулярки типа таких сюда и выставить правильный тип.
https://github.com/v-zhuravlev/zbx-smartctl/blob/master/discovery-scripts/nix/smartctl-disks-discovery.pl#L201
https://github.com/v-zhuravlev/zbx-smartctl/blob/master/discovery-scripts/windows/smartctl-disks-discovery.ps1#L105

vopatrushev, 2019-11-01:
Решение придумали - в фильтрах обнаружения поставили на ssd disktype not equal 0
Других устройств кроме hdd и ssd нет, поэтому работает корректно.

v-zhuravlev, 2019-11-01:
Спасибо за ответ.

closed 2019-11-01 by v-zhuravlev