v-zhuravlev/zbx-smartctl #135: Intel d3 s4610, hw raid hdd hpe (sas) - not work

StoryPo, 2020-04-12:
I have a hp proliant dl 360 g6 server. With 1 intel d3 4610 DC drive and two drives in Raid 0 (SAS).
Connected via the raid controller HP Smart Array P410.
I checked everything a dozen times - there are no errors. The script is executed, and the discovery rules work.
The zabbix_server.log is quiet.
At the same time, there is another server from hp with software raid - everything works perfectly.

What is the problem?

StoryPo, 2020-04-12:
`./smartctl-disks-discovery.pl 
{
	"data":[
		{
			"{#DISKMODEL}":"HP LOGICAL VOLUME",
			"{#DISKSN}":"MYSN",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d scsi",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"HP LOGICAL VOLUME",
			"{#DISKSN}":"MYSN",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d scsi",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"HP LOGICAL VOLUME",
			"{#DISKSN}":"MYSN",
			"{#DISKNAME}":"/dev/sg1",
			"{#DISKCMD}":"/dev/sg1",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"0"
		},
		{
			"{#DISKMODEL}":"HP LOGICAL VOLUME",
			"{#DISKSN}":"MYSN",
			"{#DISKNAME}":"/dev/sg2",
			"{#DISKCMD}":"/dev/sg2",
			"{#SMART_ENABLED}":"0",
			"{#DISKTYPE}":"0"
		}
	]
}`

v-zhuravlev, 2020-05-19:
SMART_ENABLED: 0 - such disks are not discovered by default template discovery rule. 
p.s. you might consider to use out of the box zabbix HP template to monitor HP servers RAID controller and disks instead.

StoryPo, 2020-05-24:
> p.s. you might consider to use out of the box zabbix HP template to monitor HP servers RAID controller and disks instead.

Is it really not necessary to monitor SMART for disks that the controller serves?
I always thought that there is a Raid controller being used, then SMART monitoring is required.

v-zhuravlev, 2020-05-24:
As far as I know, there are smart metrics in that template as well.

Отправлено с телефона

StoryPo, 2020-06-03:
@v-zhuravlev 
Hi.
I finally got to this problem.
I have a p212 + raid 10 controller.
When using `smartctl -d sat + cciss, 0 -x / dev / sda` I see smart.
If I add macros to the virtual host, then I do not see the data.
`{$ SMARTCTL_STATIC_DISKS} / dev / sda_-d_sat + cciss, 0_-x_ / dev / sda, / dev / sda_-d_sat + cciss, 1_-x_ / dev / sda, / dev / sda_-d_sat + cciss, 2_- x_ / dev / sda, / dev / sda_-d_sat + cciss, 3_-x_ / dev / sda`
`{$ SMART_ENABLED} 1`.
Am I doing something wrong?

closed 2020-06-03 by StoryPo