v-zhuravlev/zbx-smartctl #136: Ошибка в пути к smartctl

benderit, 2020-05-14:
Скрипт дискавери на перле жестко привязан к пути /usr/sbin/smartctl

в sudoers.d добавлены разрешения для юзера zabbix на /usr/sbin/smartctl

но! в /etc/zabbix/zabbix_agentd.conf.d/zabbix_smartctl.conf прописан sudo smartctl

Проблема в том, что при компилировании новой версии smartctl она находится немного по другому пути /usr/local/sbin/smartctl

поэтому проблема решается в 2 шага:
1. ln -s /usr/local/sbin/smartctl /usr/sbin/smartctl
примерно так, или копированием бинарника

2. Исправлением пути в /etc/zabbix/zabbix_agentd.conf.d/zabbix_smartctl.conf
с sudo smartctl на /usr/sbin/smartctl

damianden, 2022-04-07:
sudoers_zabbix_smartctl
в sudoers.d добавлены разрешения для юзера zabbix  указать точный путь к новому бинарнику  smartctl  , у меня **не сработало** если указывать на симлинк . Вот так не работает: ln -s /usr/local/sbin/smartctl /usr/sbin/smartctl
Для MACOS заработало вот так:
Defaults:zabbix !requiretty
Cmnd_Alias SMARTCTL = /usr/local/Cellar/smartmontools/7.3/bin/smartctl
Cmnd_Alias SMARTCTL_DISCOVERY = /etc/zabbix/scripts/smartctl-disks-discovery.pl
zabbix ALL= (ALL) NOPASSWD: SMARTCTL, SMARTCTL_DISCOVERY
Defaults!SMARTCTL !logfile, !syslog, !pam_session
Defaults!SMARTCTL_DISCOVERY !logfile, !syslog, !pam_session

benderit, 2022-04-07:
> sudoers_zabbix_smartctl в sudoers.d добавлены разрешения для юзера zabbix указать точный путь к новому бинарнику smartctl , у меня **не сработало** если указывать на симлинк . Вот так не работает: ln -s /usr/local/sbin/smartctl /usr/sbin/smartctl Для MACOS заработало вот так: Defaults:zabbix !requiretty Cmnd_Alias SMARTCTL = /usr/local/Cellar/smartmontools/7.3/bin/smartctl Cmnd_Alias SMARTCTL_DISCOVERY = /usr/local/etc/zabbix/scripts/smartctl-disks-discovery.pl zabbix ALL= (ALL) NOPASSWD: SMARTCTL, SMARTCTL_DISCOVERY Defaults!SMARTCTL !logfile, !syslog, !pam_session Defaults!SMARTCTL_DISCOVERY !logfile, !syslog, !pam_session

спасибо, тестил только на ubuntu