v-zhuravlev/zbx-smartctl #140: Criticial SSD temperature

rajil, 2020-06-25:
The critical HDD temperature threshold is set via {$TEMP_WARN:"HDD"}. It seems SSD also uses the same critical threshold. The SSD normally can go much higher than 50C. It would be nice to have a separate threshold temperature for SSD.

tabyr, 2020-07-08:
I think its not need. Because in normal DC temp will be at 22-25. At home about 40.
If you have more than 50, its time to think that cooling of PC or server its not normal.

rajil, 2020-07-10:
The SSD are vastly different to hard-disks and rated to work fine until 70C. For my system I do ensure the hard disks stay below 40 with sufficient cooling, however not so much for SSD. The SSD normally stay in 40's and occasionally go beyond 50C. 

A separate definition for SSD and HDD will be very useful.