v-zhuravlev/zbx-smartctl #154: "Unsupported item key" in Zabbix 5.0.7

fennecz1, 2021-01-21:
Hi, 

I use zbx-smartctl with zabbix 5.0.7. 
I have 2 errors, and I have no idea to how fix them.

smartctl-disks-discovery (HDD) --> Item does not allow parameters.
smartctl-disks-discovery (SSD) --> Unsupported item key.

Any ideas ? 

Thank you

rjrouquette, 2021-01-22:
Did you create and install the agent from the deb file?  If so, there's a bug in the creation of the deb file that puts zabbix_smartctl.conf in the wrong agent configuration directory.

Running the following command and restarting the agent fix the issue for me.
sudo mv /etc/zabbix/zabbix_agentd.conf.d/zabbix_smartctl.conf /etc/zabbix/zabbix_agentd.d/

fennecz1, 2021-01-22:
Hi @rjrouquette, 

I've just try it, but it doesn't work. 

``` 
root@xxxxx:/etc/zabbix# tree
.
├── scripts
│   └── smartctl-disks-discovery.pl
├── zabbix_agentd.conf
├── zabbix_agentd.conf.d
└── zabbix_agentd.d
    └── zabbix_smartctl.conf

3 directories, 3 files 
```

Here the message in the web interface 
[image]

mrPsycho, 2021-02-20:
+ triggers are not working on 5