v-zhuravlev/zbx-smartctl #162: Zabbix agent does not sending data - Unsupported item key

liashuk-se, 2021-04-12:
Cannot fetch data from server with 2 attached directly disks from zabbix server. 

Running perl discovery script itself on monitored node as a zabbix user returns data correctly 
`# sudo -u zabbix sudo /etc/zabbix/scripts/smartctl-disks-discovery.pl

{
	"data":[
		{
			"{#DISKMODEL}":"SAMSUNG MZ7LM480HMHQ-00005",
			"{#DISKSN}":"S2UJNX0M502684",
			"{#DISKNAME}":"/dev/sda",
			"{#DISKCMD}":"/dev/sda -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		},
		{
			"{#DISKMODEL}":"SAMSUNG MZ7LM480HMHQ-00005",
			"{#DISKSN}":"S2UJNX0M506845",
			"{#DISKNAME}":"/dev/sdb",
			"{#DISKCMD}":"/dev/sdb -d sat",
			"{#SMART_ENABLED}":"1",
			"{#DISKTYPE}":"1"
		}
	]
}`

Attempt to fetch same data from zabbix server 
`zabbix_get -s 172.16.101.50 -p 10050 -k uSSD.discovery[""]
ZBX_NOTSUPPORTED: Unsupported item key.`

From monitored server zabbix-agent logs (debug level):
` 10829:20210412:134723.739 Requested [uSSD.discovery[]]
 10829:20210412:134723.739 Sending back [ZBX_NOTSUPPORTED: Unsupported item key.]`

Zabbix server version 4.2.8

Montiored server details:

- Zabbix-agent version 4.4.10
- Centos 7
- smartmontools 7.0-2.el7

dazgluk, 2021-05-01:
Having the same issue on Ubuntu 20.04 LTS and Zabbix 4.4.6

dazgluk, 2021-05-03:
Fixed an issue for my self, there was a permission issue, zabbix user had no access to newly created /etc/zabbix/conf.d/ file

proteus88, 2021-08-20:
Had the same problem "Unsupported item key", the zabbix agent config file did not know about the zabbix_smartctl.conf file, I had to add this line
Include=/etc/zabbix/zabbix_agentd.d/*.conf

tommer-cpu, 2024-10-04:
> Had the same problem "Unsupported item key", the zabbix agent config file did not know about the zabbix_smartctl.conf file, I had to add this line Include=/etc/zabbix/zabbix_agentd.d/*.conf

 Всем привет!У меня вот  какая проблема 
[image]

tommer-cpu, 2024-10-04:
> If you use Zabbix Agent 2 you don't need this template/script. Just use the official template from Zabbix and "sudo apt install smartmontools" then add "zabbix ALL=(ALL) NOPASSWD:/usr/sbin/smartctl" with "sudo visudo" command.
> 
> https://www.zabbix.com/integrations/smart

А если это windows zabbix agent?

tommer-cpu, 2024-10-07:
[image]
Now I have this